wolfSSL/wolfssl #2911: USE of SHA3-HMAC !

bosssaurav, 2020-04-19:
Hi,

 I want to make use of SHA3-HMAC. So far as I have understood, I think SHA3 implementation is blake2b. Can it be used directly to HMAC API ? 

Could you provide a small piece of code as how to achieve this  ? 

Also with the required header file to be included along with so's.

Also does wolfssl have implementation for merkle tree ? 

Kindly help. :)

dgarske, 2020-04-20:
Hi @bosssaurav ,

SHA-3 is not Blake2. Blake2 was one of the finalists for the SHA-3 specification, but the final algorithm for SHA3 was Keccak. See https://en.wikipedia.org/wiki/NIST_hash_function_competition

Here is a simple SHA-3 256 HMAC example:

```
/* Example for HMAC SHA3-256 */
/* gcc -lwolfssl -o hmac hmac_sha3_256_test.c */

#ifndef WOLFSSL_USER_SETTINGS
    #include <wolfssl/options.h>
#endif
#include <wolfssl/wolfcrypt/settings.h>
#include <wolfssl/wolfcrypt/hmac.h>

#include <stdio.h>

#if !defined(NO_HMAC) && defined(WOLFSSL_SHA3)
int hmac_sha3_256_test(void)
{
    int ret;
    Hmac hmac;
    byte hash[WC_SHA3_256_DIGEST_SIZE];

    const char* key = "Jefe";
    const char* input  = "what do ya want for nothing?";
    const char* output = 
        "\xc7\xd4\x07\x2e\x78\x88\x77\xae\x35\x96\xbb\xb0\xda\x73\xb8\x87"
        "\xc9\x17\x1f\x93\x09\x5b\x29\x4a\xe8\x57\xfb\xe2\x64\x5e\x1b\xa5";
    word32 inLen  = XSTRLEN(input);
    word32 outLen = WC_SHA3_256_DIGEST_SIZE;

    printf("HMAC SHA3-256 Test Vector\n");

    ret = wc_HmacInit(&hmac, NULL, INVALID_DEVID);
    if (ret != 0) {
        return ret;
    }

    /* Setup HMAC key */
    ret = wc_HmacSetKey(&hmac, WC_SHA3_256, (byte*)key, (word32)XSTRLEN(key));
    if (ret == 0) {
        /* Update HMAC with data */
        ret = wc_HmacUpdate(&hmac, (byte*)input, (word32)inLen);
    }
    /* Get final HMAC digest */
    if (ret == 0) {
        ret = wc_HmacFinal(&hmac, hash);
    }

    /* confirm output test vector */
    if (ret == 0 && XMEMCMP(hash, output, WC_SHA3_256_DIGEST_SIZE) != 0) {
        ret = -1;
    }

    wc_HmacFree(&hmac);

    printf("Test Result: %d\n", ret);

    return ret;
}
#endif

int main()
{
#if !defined(NO_HMAC) && defined(WOLFSSL_SHA3)
    return hmac_sha3_256_test();
#else
    printf("Not compiled in!\n");
    return -1;
#endif   
}
```

Thanks,
David Garske, wolfSSL

bosssaurav, 2020-04-21:
Hi @dgarske

Thanks for quick response. 
wc_HmacInit(&hmac, NULL, INVALID_DEVID);

**what does parameter  INVALID_DEVID ?**

also to my second question : **Also does wolfssl have implementation for merkle tree ?**

dgarske, 2020-04-21:
Hi @bosssaurav , 

When a "devId" is provided it means use a cryptographic hardware device identifier. This is used for our asynchronous version of the wolfSSL library, which can utilize the Intel QuickAssist or Cavium Nitrox. The devId can also be used to register a crypto callback using our `./configure --enable-cryptocb` or `#define WOLF_CRYPTO_CB` feature. https://www.wolfssl.com/wolfcrypt-support-cryptographic-callbacks/

The `#define INVALID_DEVID -2` is a value to indicate no device should be used.
No we don't have a support for a merkle tree. Wiki says "Merkle tree is a tree in which every leaf node is labelled with the cryptographic hash of a data block, and every non-leaf node is labelled with the cryptographic hash in the labels of its child nodes." That's an interesting feature request. Can you tell me more about the use-case you are interested in? Feel free to send us a note directly support@wolfssl.com reguarding this feature request.

Thanks,
David Garske, wolfSSL

bosssaurav, 2020-04-23:
Thanks David.. vl get in touch with you
For now lets close this one :)

closed 2020-04-23 by bosssaurav