wolfSSL/wolfssl #2924: mp_exptmod error state -112 at wolfcrypt/src/rsa.c:2165

grooverdan, 2020-04-25:
wolfssl at ebb490204afdd1b76ba2ac732f94143b84143851
certs per https://github.com/MariaDB/server/tree/10.4/mysql-test/std_data

x86_64 compile:
` ./configure --enable-tls13 --enable-curve25519  --enable-hkdf --enable-maxstrength`

Run server:

` examples/server/server -c /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/server8k-cert.pem -k /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/server8k-key.pem -A /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/cacert.pem -i -d -v 4`

client connection either gnutls or openssl:

`gnutls-cli --no-ca-verification  --x509keyfile=client-certkey.pem --x509certfile=client-cert.pem -p 11111 localhost -d 5`

<pre>
openssl s_client -tls1_3  -connect localhost:11111  -cert client-cert.pem -key client-certkey.pem -debug
..
---
SSL handshake has read 1894 bytes and written 217 bytes
Verification error: unable to verify the first certificate
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 8192 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 21 (unable to verify the first certificate)
</pre>

results in the server error output:
```

SSL_accept error -112, mp_exptmod error state
wolfSSL error: SSL_accept failed
```

equalivant openssl s_server succesfully to forms a connection:
`
openssl s_server -tls1_3  -cert  /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/server8k-cert.pem -key  /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/server8k-key.pem -CAfile  /home/dan/repos/mariadb-server-10.4/mysql-test/std_data/cacert.pem  -port 11111`

gdb of server:
break wc_RsaFunctionSync
r
(run client side and hits breakpoint)
watch ret
c
c
(gets -112 error)

Backtrace
<pre>
Old value = 0
New value = -112
wc_RsaFunctionSync (
    in=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., inLen=1024, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=0x411da8, type=2, key=0x40fb50, rng=0x40b010) at wolfcrypt/src/rsa.c:2165
2165	            if (ret == 0 && mp_mulmod(tmp, rnd, &key->n, tmp) != MP_OKAY)
(gdb) bt
#0  wc_RsaFunctionSync (
    in=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., inLen=1024, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=0x411da8, type=2, key=0x40fb50, rng=0x40b010) at wolfcrypt/src/rsa.c:2165
#1  0x00007ffff7f6d569 in wc_RsaFunction (
    in=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., inLen=1024, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=0x411da8, type=2, key=0x40fb50, rng=0x40b010) at wolfcrypt/src/rsa.c:2660
#2  0x00007ffff7f6d726 in RsaPublicEncryptEx (
    in=0x40cfa0 "\330\376\246\317we:\335\346ؓk\336\323\310̧\371\257\273\037И1\f\215\301\b9\343~=", inLen=32, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=1024, key=0x40fb50, rsa_type=2, pad_value=1 '\001', pad_type=2, hash=WC_HASH_TYPE_SHA256, mgf=1, 
    label=0x0, labelSz=0, saltLen=-1, rng=0x40b010) at wolfcrypt/src/rsa.c:2781
#3  0x00007ffff7f6e514 in wc_RsaPSS_Sign_ex (
    in=0x40cfa0 "\330\376\246\317we:\335\346ؓk\336\323\310̧\371\257\273\037И1\f\215\301\b9\343~=", inLen=32, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=1024, hash=WC_HASH_TYPE_SHA256, mgf=1, saltLen=-1, key=0x40fb50, rng=0x40b010)
    at wolfcrypt/src/rsa.c:3525
#4  0x00007ffff7f6e4b9 in wc_RsaPSS_Sign (
--Type <RET> for more, q to quit, c to continue without paging--
    in=0x40cfa0 "\330\376\246\317we:\335\346ؓk\336\323\310̧\371\257\273\037И1\f\215\301\b9\343~=", inLen=32, 
    out=0x40d5bd "N\n\346\252=%\360\301\065'\026\216\030\311_\256\272\021Or8hL\031\361\375\257%\311\"1\005\221\252\336\357f\342#\204\262\271\200\355\360?V\f\004\263\204C\361\367)];\025\niԃk\244\216\244\270\355\061V\275\334đ\252˥\215O,6Uk\003\343&\204\265&\252/ڨxl\347\355\367*6\237\r\211>\\R\v|e3\230c-\321b\277\233}\357\023\r\245*ܧ\200E9\001\375\273N\217h-\201\323\352\071\r\262\240F\207\330\321\022n\234\351\v=nmD\215\304+\264\366\373|?- ɹ\244\213bq\217)\244\f\327%\361\274G\033V\234\224C\020\206\201o\353=\201\035[\261\071\332", <incomplete sequence \353\276>..., 
    outLen=1024, hash=WC_HASH_TYPE_SHA256, mgf=1, key=0x40fb50, rng=0x40b010) at wolfcrypt/src/rsa.c:3501
#5  0x00007ffff7f909a2 in RsaSign (ssl=ssl@entry=0x40a6b0, in=<optimized out>, inSz=<optimized out>, 
[... cut ...]
    keyBufInfo=0x40e280) at src/internal.c:3687
#6  0x00007ffff7faded4 in SendTls13CertificateVerify (ssl=ssl@entry=0x40a6b0) at src/tls13.c:5795
#7  0x00007ffff7fb0d36 in wolfSSL_accept_TLSv13 (ssl=ssl@entry=0x40a6b0) at src/tls13.c:8788
#8  0x00007ffff7fa48a6 in wolfSSL_accept (ssl=ssl@entry=0x40a6b0) at src/ssl.c:11971
#9  0x0000000000403cb3 in server_test (args=args@entry=0x7fffffffdae0) at examples/server/server.c:2236
#10 0x0000000000402586 in main (argc=<optimized out>, argv=<optimized out>) at examples/server/server.c:2555
</pre>

dgarske, 2020-04-27:
Hi @grooverdan ,

The -112 error is a math precision error.

For 8K RSA keys you'll need to increase the Multiple Precision (MP) math `FP_MAX_BITS` to max key size * 2 (`8192 * 2 = 16384`).

Try using the following configure:
`./configure --enable-tls13 --enable-curve25519 --enable-hkdf --enable-maxstrength CFLAGS="-DFP_MAX_BITS=16384`.

Thanks,
David Garske, wolfSSL

grooverdan, 2020-04-27:
great. That got me to an isolated reproducable case for #2904. I was assuming maxstrength implied the FP_MAX_BITS. My bad. Thanks.

closed 2020-04-27 by grooverdan