wolfSSL/wolfssl #2949: Allow ASN_IP_TYPE without enabling QT or OpenSSL support

elagergren-spideroak, 2020-05-05:
Right now, `ASN_IP_TYPE` support is hidden behind the QT and OpenSSL feature flags: https://github.com/wolfSSL/wolfssl/blob/5e45767cc30d0edba18685f8badc264a7ef761bd/wolfcrypt/src/asn.c#L7802

We need to support TLS certificates that use the `ipAddress` type as defined in [RFC 5280](https://tools.ietf.org/html/rfc5280#section-4.2.1.6), but we do not want to enable QT or OpenSSL support.

Additionally, enabling QT support requires we enable DH and TLS <= v1.2, as evidenced by compiler errors when `--enable-qt` is combined with `--disable-dh` and `--enable-maxstrength`:

```
Undefined symbols for architecture x86_64:
  "_MakeSSLv3", referenced from:
      _wolfSSLv3_client_method_ex in libwolfssl.a(libwolfssl_la-ssl.o)
      _wolfSSLv3_server_method_ex in libwolfssl.a(libwolfssl_la-ssl.o)
Undefined symbols for architecture x86_64:
  "_MakeSSLv3", referenced from:
      _wolfSSLv3_client_method_ex in libwolfssl.a(libwolfssl_la-ssl.o)
      _wolfSSLv3_server_method_ex in libwolfssl.a(libwolfssl_la-ssl.o)
  "_wolfSSL_EVP_PKEY_assign_DH", referenced from:
      _wolfSSL_EVP_PKEY_assign in libwolfssl.a(libwolfssl_la-ssl.o)
  "_wolfSSL_EVP_PKEY_assign_DH", referenced from:
      _wolfSSL_EVP_PKEY_assign in libwolfssl.a(libwolfssl_la-ssl.o)
```

julek-wolfssl, 2020-05-18:
This functionality has been implemented in #2958. Closing.

closed 2020-05-18 by julek-wolfssl

elagergren-spideroak, 2020-05-22:
@julek-wolfssl thank you!