wolfSSL/wolfssl #2951: wolfcrypt/hkdf: does not conform to RFC 5869

elagergren-spideroak, 2020-05-06:
[RFC 5869 section 2.3](https://tools.ietf.org/html/rfc5869#section-2.3) states that the length of the output is `L <= 255*HashLen` where `HashLen` is the size in octets of the the hash function's digest. However, `wc_HKDF_Expand` has no such limit: https://github.com/wolfSSL/wolfssl/blob/5e45767cc30d0edba18685f8badc264a7ef761bd/wolfcrypt/src/hmac.c#L1209-L1252

Examples from other HKDF libraries:

1. Go: https://github.com/golang/crypto/blob/4b2356b1ed79e6be3deca3737a3db3d132d2847a/hkdf/hkdf.go#L49
2. OpenSSL: https://github.com/OneSignal/openssl/blob/0db957dbbcf6a432086ab913378c23636d8c374c/crypto/kdf/hkdf.c#L290-L297
3. Rust: https://github.com/RustCrypto/KDFs/blob/6c61c600ec03f02d36a301f320fe6a957711e6a7/hkdf/src/hkdf.rs#L108
4. Python: https://github.com/pyca/cryptography/blob/e05795897c7b723e134da5085fd60fa710fb5081/src/cryptography/hazmat/primitives/kdf/hkdf.py#L67-L73

JacobBarthelmeh, 2020-07-07:
Closing this out with the fix in PR https://github.com/wolfSSL/wolfssl/pull/2956 having been merged.

JacobBarthelmeh, 2020-07-07:
Thanks @elagergren-spideroak for the report

closed 2020-07-07 by JacobBarthelmeh