wolfSSL/wolfssl #2960: TLSv1.0 broken in 4.4.0 (tested with schannel)

vaintroub, 2020-05-08:
Testing MariaDB, found that e6affa386f67cb6428d95bdf0d3de1adb78dfb79 breaks TLSv1.0 with Schannel.
[network_trace.zip](https://github.com/wolfSSL/wolfssl/files/4596331/network_trace.zip)

vaintroub, 2020-05-08:
Found via git bisect

vaintroub, 2020-05-08:
Perhaps, Schannel is confused, since a server which does not support TLSv1.3  sends TLSv1.3 "DOWNGRD" in its "random"

We build so far without 1.3 support, and in this case also instructed WolfSSL server to use TLSv1.0 only.

SparkiDev, 2020-05-11:
Hi @vaintroub,

I believe I've got a fix for this: https://github.com/wolfSSL/wolfssl/pull/2964
The problem appears to be that wolfSSL wasn't checking the flag for no TLS v1.2.

Please  let me know if this works!

Sean

vaintroub, 2020-05-11:
@SparkiDev , yes, this patch works for me.

SparkiDev, 2020-05-14:
Hi @vaintroub,

Thanks for reporting the issue!
The pull request has been merged and the patch worked for you so I'm closing this issue.

Sean

closed 2020-05-14 by SparkiDev