wolfSSL/wolfssl #2976: Cross-compiling with clang with --enable-curve448 fails

guidovranken, 2020-05-17:
Cross-compiling on 64 bit for 32 bit on Linux.

Build procedure:

```
export CC=clang
export CFLAGS="-m32"
setarch i386 ./configure --enable-curve448
make
```

Output:

```
wolfcrypt/src/fe_448.c:1781:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t0  >> 28; t1  += o; t = (int64_t)o << 28; t0  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1782:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t1  >> 28; t2  += o; t = (int64_t)o << 28; t1  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1783:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t2  >> 28; t3  += o; t = (int64_t)o << 28; t2  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1784:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t3  >> 28; t4  += o; t = (int64_t)o << 28; t3  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1785:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t4  >> 28; t5  += o; t = (int64_t)o << 28; t4  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1786:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t5  >> 28; t6  += o; t = (int64_t)o << 28; t5  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1787:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t6  >> 28; t7  += o; t = (int64_t)o << 28; t6  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1788:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t7  >> 28; t8  += o; t = (int64_t)o << 28; t7  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1789:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t8  >> 28; t9  += o; t = (int64_t)o << 28; t8  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1790:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t9  >> 28; t10 += o; t = (int64_t)o << 28; t9  -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1791:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t10 >> 28; t11 += o; t = (int64_t)o << 28; t10 -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1792:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t11 >> 28; t12 += o; t = (int64_t)o << 28; t11 -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1793:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t12 >> 28; t13 += o; t = (int64_t)o << 28; t12 -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1794:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t13 >> 28; t14 += o; t = (int64_t)o << 28; t13 -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1795:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t14 >> 28; t15 += o; t = (int64_t)o << 28; t14 -= t;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1796:13: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    o = t15 >> 28; t0  += o;
      ~ ~~~~^~~~~
wolfcrypt/src/fe_448.c:1800:12: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    r[0] = t0;
         ~ ^~
wolfcrypt/src/fe_448.c:1801:12: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    r[1] = t1;
         ~ ^~
wolfcrypt/src/fe_448.c:1802:12: error: implicit conversion loses integer precision: 'int64_t' (aka 'long long') to 'int32_t' (aka 'int') [-Werror,-Wshorten-64-to-32]
    r[2] = t2;
         ~ ^~
```

SparkiDev, 2020-05-17:
Hi Guido,

I've prepared a fix for this: https://github.com/wolfSSL/wolfssl/pull/2978
Let me know if I missed something!

Thanks,
Sean

guidovranken, 2020-05-18:
Thank you, confirmed fixed.

closed 2020-05-18 by guidovranken