wolfSSL/wolfssl #2999: --enable-all options compiling error

sacca97, 2020-05-26:
[image]

Both master and latest release.

JacobBarthelmeh, 2020-05-26:
Hi @sacca97. Thanks for the report. Which compiler can be used to reproduce this? So far have tried on my local machine with gcc 9.2.0 and an older version of clang without success in reproducing the compiler message.

sacca97, 2020-05-26:
@JacobBarthelmeh I'm on gcc 10.1.0, didn't think about trying an earlier version though.

JacobBarthelmeh, 2020-05-27:
I can reproduce the compiler error with gcc 10.1 and am working on a solution. A work around is using older compiler version.

JacobBarthelmeh, 2020-05-29:
Hi @sacca97 a fix for this was merged into master.

closed 2020-05-29 by JacobBarthelmeh