wolfSSL/wolfssl #3000: Version 4.4.0 wolfSSL_RSA_public_decrypt Undefined

zero-rp, 2020-05-27:
On Windows platform

kaleb-himes, 2020-06-02:
Hi @zero-rp,

Can you send me the wolfssl-4.4.0/IDE/WIN/user_settings.h you used to produce this issue so I can reproduce and fix?

Thanks!

- K

zero-rp, 2020-06-03:
#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_

#define WC_RSA_BLINDING

#define OPENSSL_ALL
#define OPENSSL_EXTRA

#define WOLFSSL_RIPEMD
#define NO_PSK
#define HAVE_EXTENDED_MASTER
#define HAVE_TLS_EXTENSIONS
#define HAVE_ALPN
#define WOLFSSL_ALWAYS_KEEP_SNI
#define WOLFSSL_ALLOW_TLSV10
#define WOLFSSL_TLS13
#define HAVE_AES_ECB
#define HAVE_AES_KEYWRAP
#define HAVE_AESGCM
#define HAVE_AESCCM
#define WOLFSSL_AES_CFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_XTS
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define HAVE_ECC
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_KOBLITZ
#define HAVE_ECC_SECPR2
#define HAVE_ECC_BRAINPOOL
#define HAVE_ECC_SECPR3
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define WOLFSSL_CUSTOM_CURVES
#define HAVE_SUPPORTED_CURVES
#define HAVE_SECURE_RENEGOTIATION
#define WC_NO_HARDEN
#define WOLFSSL_KEY_GEN
#define HAVE_COMP_KEY
#define WOLFSSL_DES_ECB
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
#define HAVE_LIBZ
#define HAVE_PKCS7
#define HAVE_HKDF
#define HAVE_X963_KDF
#define WOLFSSL_CERT_EXT
#define WOLFSSL_CERT_GEN
#define WOLFSSL_CERT_REQ
#define WOLFSSL_DER_LOAD
#define WOLFSSL_DTLS
#define HAVE_CAMELLIA
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_ONE_TIME_AUTH
#define HAVE_CRL
#define HAVE_CRL_IO
#define HAVE_OCSP
#define WOLFSSL_SNIFFER
#define WOLFSSL_SNIFFER_WATCH
#define WOLFSSL_SNIFFER_STATS
#define HAVE_CURVE25519

/* Optional Performance Speedups */
#if 1
    /* AESNI on x64 */
#if(defined(_WIN64) || defined(USE_ARCH_X86_64) || defined(USE_ARCH_AMD64))
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#endif

/* Single Precision Support for RSA/DH 1024/2048/3072 and ECC P-256 */
#define WOLFSSL_SP
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_RSA
#endif

#if _DEBUG
#define DEBUG_WOLFSSL
#endif

#endif /* _WIN_USER_SETTINGS_H_ */

zero-rp, 2020-06-03:
wolfssl\src\ssl.c   line:44677

#if !defined(_WIN32) && !defined(HAVE_FIPS) && !defined(HAVE_SELFTEST)
int wolfSSL_RSA_public_decrypt(int flen, const unsigned char* from,

kaleb-himes, 2022-11-23:
This issue was just brought to my attention by a colleague. I must apologize for the long delay in responding and feel I should include the history before closing it out as resolved.

When the report came in it had already been fixed post-release and was not reproducible any longer.
The user_settings.h did have a slight issue in it so I'm including the modified version that was used in the attempt to reproduce the report for posterity:

```
#ifndef WIN_USER_SETTINGS_H
#define WIN_USER_SETTINGS_H

#define WC_RSA_BLINDING

#define OPENSSL_ALL
#define OPENSSL_EXTRA

#define WOLFSSL_RIPEMD
#define NO_PSK
#define HAVE_EXTENDED_MASTER
#define HAVE_TLS_EXTENSIONS
#define HAVE_ALPN
#define WOLFSSL_ALWAYS_KEEP_SNI
#define WOLFSSL_ALLOW_TLSV10
#define WOLFSSL_TLS13
#define HAVE_AES_ECB
#define HAVE_AES_KEYWRAP
#define HAVE_AESGCM
#define HAVE_AESCCM
#define WOLFSSL_AES_CFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_XTS
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define HAVE_ECC
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_KOBLITZ
#define HAVE_ECC_SECPR2
#define HAVE_ECC_BRAINPOOL
#define HAVE_ECC_SECPR3
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define WOLFSSL_CUSTOM_CURVES
#define HAVE_SUPPORTED_CURVES
#define HAVE_SECURE_RENEGOTIATION
#define WC_NO_HARDEN
#define WOLFSSL_KEY_GEN
#define HAVE_COMP_KEY
#define WOLFSSL_DES_ECB
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
//#define HAVE_LIBZ <--- LIBZ is for compression but wasn't associated with the issue reported. Removed as the windows system being tested on did not have libz available.
#define HAVE_PKCS7
#define HAVE_HKDF
#define HAVE_X963_KDF
#define WOLFSSL_CERT_EXT
#define WOLFSSL_CERT_GEN
#define WOLFSSL_CERT_REQ
#define WOLFSSL_DER_LOAD
#define WOLFSSL_DTLS
#define HAVE_CAMELLIA
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_ONE_TIME_AUTH
#define HAVE_CRL
#define HAVE_CRL_IO
#define HAVE_OCSP
#define WOLFSSL_SNIFFER
#define WOLFSSL_SNIFFER_WATCH
#define WOLFSSL_SNIFFER_STATS
#define HAVE_CURVE25519

/* Optional Performance Speedups /
#if 1
/ AESNI on x64 */
#if defined(_WIN64) || defined(USE_ARCH_X86_64) || defined(USE_ARCH_AMD64) // <--- MSVS 2019 didn't like the syntax on this line in the original (see github issue #3000)
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#endif

/* Single Precision Support for RSA/DH 1024/2048/3072 and ECC P-256 */
#define WOLFSSL_SP
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_RSA
//#endif <-- Erroneous endif removed

#if _DEBUG
#define DEBUG_WOLFSSL
#endif

#endif /* WIN_USER_SETTINGS_H */
```

closed 2022-11-23 by kaleb-himes