wolfSSL/wolfssl #3006: ExtractDate(asn.h) for year 2020 returns tm_year 20, not 120 as expected

lstrizu, 2020-05-28:
Made a small test on ExtractDate, by providing the following date byte array:
byte dateArray2[]{0, 0, 50, 48, 48, 53, 50, 56, 49, 50, 49, 49, 49, 49, 90}; //date 2020.05.28, 12:11:11 UTC
and calling:
ExtractDate(dateArray2, ASN_UTC_TIME, &tmpTm, &idx);

I expect that tmpTm.tm_year to be 120 (years since 1900), but  it is 20.
Am I missing something?

dgarske, 2020-05-28:
Hi @Istrizu,

Looks like it uses 1900 here:
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L6126
Then subtracts 1900 here:
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L6137

For it to return 120 this line of code `if (btoi(date[*idx]) >= 5)` to evaluate false so 2000 is used. Is your `&idx` parameter 4 initialized correctly?

Thanks,
David Garske, wolfSSL

lstrizu, 2020-05-29:
Hi,

Thanks for answering.
So, idx is set to 2, as my dateArray2 has the first 2 entries 0.
So  ExtractDate will start processing the rest of the array: 50, 48, 48, 53, 50, 56, 49, 50, 49, 49, 49, 49, 90 ( which is the expected date in UTC 200528121111Z).
(btoi(date[*idx]) >= 5) will translate into (btoi(50) >= 5) which I assume it will return TRUE, thus adding 1900 <- here I think is not correct, as 50, 48 (20) denotes year 2020, thus it should add 2000, and later subtract 1900 , which will results in correct 120 
Maybe I don't provide the correct date array. If so, could you please give me an example of a correct one? 

Thanks,
Laura Strizu

closed 2021-01-15 by haydenroche5