wolfSSL/wolfssl #3014: Split AAD & GCM Encrypt steps with wc_AesGcmEncrypt

bryan-hunt, 2020-05-31:
Based on my reading of wc_AesGcmEncrypt I should be able to perform a separate AAD step from the encryption step but that doesn't seem to be working for me so hopefully you might have some insights on how to best resolve this.

Version: v4.3.0-stable

I'm using the McGrew & Viega test vectors for verifying the following implementations. I've tried many attempts at getting WolfCrypt but I can't seem to get it to produce the proper tags.. When I only make one call to wc_AesGcmEncrypt with aad then the tag produced will be correct. Unfortunately combining the AAD with the Encryption is not desirable in the system I am working on (essentially caching the AAD until the encrypt/decrypt step). The OpenSSL compatibility layer isn't available to me in the target environment.

WolfSSL:
```
Setup Encryption
wc_AesGcmSetKey(&ctx->aes, key, key_len);

Update AAD:
wc_AesGcmEncrypt(&ctx->aes, NULL, NULL, 0, iv, iv_len, tag, tag_len, aad, (int)aad_len);

Encrypt Data:
wc_AesGcmEncrypt(&ctx->aes, ciphertext, plaintext, pt_len, iv, iv_len, tag, tag_len, NULL, 0);

Finish:
ciphertext & tag from above
```

For reference I also have the following in my system and validated

OpenSSL:
```
Setup Encryption
EVP_EncryptInit_ex((EVP_CIPHER_CTX*)ctx->ptr, EVP_aes_128_gcm(), NULL, NULL, NULL);
EVP_CIPHER_CTX_ctrl((EVP_CIPHER_CTX*)ctx->ptr, EVP_CTRL_AEAD_SET_IVLEN, (int)iv_len, NULL);
EVP_EncryptInit_ex((EVP_CIPHER_CTX*)ctx->ptr, NULL, NULL, key, iv);

Update AAD
EVP_CipherUpdate((EVP_CIPHER_CTX*)ctx->ptr, NULL, &outlen, aad, (int)aad_len))

Encrypt data
EVP_EncryptUpdate((EVP_CIPHER_CTX*)ctx->ptr, ciphertext, &outlen, plaintext, (int)pt_len))

Finish Encryption & Get the tag
EVP_EncryptFinal_ex((EVP_CIPHER_CTX*)ctx->ptr, NULL, &outlen);
EVP_CIPHER_CTX_ctrl((EVP_CIPHER_CTX*)ctx->ptr, EVP_CTRL_AEAD_GET_TAG, (int)tag_len, tag);
```

mBedTLS:
```
Setup Encryption:
mbedtls_cipher_setup(ctx, mbedtls_cipher_info_from_type(MBEDTLS_CIPHER_AES_128_GCM));
mbedtls_cipher_setkey(ctx, key, key_len * 8, MBEDTLS_ENCRYPT);
mbedtls_cipher_set_iv(ctx, iv, iv_len);
mbedtls_cipher_reset(ctx);

Update AAD:
mbedtls_cipher_update_ad(ctx, aad, aad_len);

Encrypt Data:
mbedtls_cipher_update(ctx, plaintext, pt_len, ciphertext, ct_len);

Finish Encryption & Get the tag:
mbedtls_cipher_finish(ctx, NULL, &outlen);
mbedtls_cipher_write_tag(ctx, tag, tag_len);
```

SparkiDev, 2020-06-02:
Hi Bryan,

The wolfCrypt library does not support a streaming interface for AES-GCM.
That is, you must call wc_AesGcmEncrypt() with all data in one shot.

If this is something you need, please let us know and we can discuss how this feature can be added to wolfSSL.

Thanks,
Sean

bryan-hunt, 2020-06-03:
That would then indicate that the GCM EVP doesn't work either.

SparkiDev, 2020-06-03:
Looking at the latest EVP wrapper code, I believe that we keep a copy of the data if there is no output buffer passed in.

I have to agree that caching the AAD is the only option.

Sean

bryan-hunt, 2020-06-03:
Thanks for looking into it. I think there is already another request for a "streaming" version of GCM where the data parts are not cached so I'll close this as a dup.

closed 2020-06-03 by bryan-hunt