wolfSSL/wolfssl #3046: I get "consider using harden options" warning with --enable-all using OpenSSL API

bagder, 2020-06-15:
Hi,

We're working on adding NTLM support for curl using wolfSSL, which has been lacking for a while (see https://github.com/curl/curl/pull/5556).

To enable this, we use the OpenSSL API for the necessary `DES_*` functions. We have previously not used the OpenSSL API layer, and granted it is a bit of a lazy shortcut but still, this is what we have here before us.

We detect if one of the necessary functions is present in configure, and it so, we add another include path to point to the wolfssl/openssl include path.

In our CI job, we build wolfssl 4.4.0 with `./configure --enable-tls13 --enable-all`. This has worked fine for quite some time. **Until I include an openssl header** in this PR... as now I get a warning that I don't have built wolfssl with hardening options. Aren't they enabled with enable-all already? I didn't get this warning until I try the openssl header.

Is this is a bug? Is there something else I should do to avoid this warning? I also get the same one with wolfssl from the master branch as of just now.

This is the exact here I get. It can be seen in our CI [here](https://travis-ci.org/github/curl/curl/jobs/698572965#L1562) which also includes the wolfssl build itself if you scroll up.
~~~
/usr/local/include/wolfssl/wolfcrypt/settings.h:2060:14: error: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Werror=cpp]
             #warning "For timing resistance / side-channel attack prevention consider using harden options"
              ^~~~~~~

dgarske, 2020-06-15:
Hi @bagder , that usually means you aren't including `#include <wolfssl/options.h>` in on your side before including any other wolfSSL headers.

bagder, 2020-06-15:
Easy, enough. Thanks, will check it out at once!

closed 2020-06-15 by bagder