wolfSSL/wolfssl #3059: CryptoCell 310 Port RSA/ECC tests failing

SylwBar, 2020-06-18:
Hi.
After applying procedure described here:
https://github.com/wolfSSL/wolfssl/tree/master/IDE/CRYPTOCELL
I got wrong output on nRF52840DK board:
```
wolfCrypt Test Started
------------------------------------------------------------------------------
 wolfSSL version 4.4.1
------------------------------------------------------------------------------
...
RSA      test failed!
 error = -7463
```
After short research I found that following commit broke successful RSA test execution on nRF52:
https://github.com/wolfSSL/wolfssl/pull/2754
Problem:
RSA_TEST_BYTES was 256, but it was changed to 384 and later to 512.
test.c:11073 #define RSA_TEST_BYTES 512 
Change:
#define RSA_TEST_BYTES 256
solves problem with RSA test. I guess this value should be parametrized. 

Next problem is in ECC test:

```
wolfCrypt Test Started
------------------------------------------------------------------------------
 wolfSSL version 4.4.1
------------------------------------------------------------------------------
...
RSA      test passed!
ecc_test_make_pub failed!: -9628
ECC      test failed!
 error = -9628
 
```
Related change was introduced by commit:
https://github.com/wolfSSL/wolfssl/pull/2738
The change adds return value check in  wolfcrypt/test/test.c:18358
```
ret  = wc_ecc_make_key(rng, 32, &key);
    if (ret == 0) {
        ERROR_OUT(-9628, done);
    }
```
I guess this check should be opposite: if (ret != 0)

After those changes applied tests and benchmarks are executed correctly:
```
wolfCrypt Test Started
------------------------------------------------------------------------------
 wolfSSL version 4.4.1
------------------------------------------------------------------------------
error    test passed!
MEMORY   test passed!
base64   test passed!
asn      test passed!
RANDOM   test passed!
SHA      test passed!
SHA-256  test passed!
Hash     test passed!
HMAC-SHA test passed!
HMAC-SHA256 test passed!
AES      test passed!
RSA      test passed!
ECC      test passed!
ECC buffer test passed!
logging  test passed!
mutex    test passed!
Test complete

wolfCrypt Test Completed

Benchmark Test Started
------------------------------------------------------------------------------
 wolfSSL version 4.4.1
------------------------------------------------------------------------------
wolfCrypt Benchmark (block bytes 1024, min 1.0 sec each)
RNG                  5 MB took 1.004 seconds,    4.815 MB/s
AES-128-CBC-enc     17 MB took 1.000 seconds,   17.456 MB/s
AES-128-CBC-dec     17 MB took 1.001 seconds,   17.414 MB/s
SHA                425 KB took 1.022 seconds,  415.851 KB/s
SHA-256             27 MB took 1.000 seconds,   26.611 MB/s
HMAC-SHA           425 KB took 1.031 seconds,  412.221 KB/s
HMAC-SHA256         24 MB took 1.000 seconds,   24.365 MB/s
RSA     1024 key gen         1 ops took 1.058 sec, avg 1058.000 ms, 0.945 ops/sec
RSA     2048 key gen         1 ops took 3.400 sec, avg 3400.000 ms, 0.294 ops/sec
RSA     2048 public        330 ops took 1.000 sec, avg 3.030 ms, 330.000 ops/sec
RSA     2048 private         4 ops took 1.694 sec, avg 423.500 ms, 2.361 ops/sec
ECC      256 key gen        55 ops took 1.006 sec, avg 18.291 ms, 54.672 ops/sec
ECDHE    256 agree          56 ops took 1.007 sec, avg 17.982 ms, 55.611 ops/sec
ECDSA    256 sign           52 ops took 1.034 sec, avg 19.885 ms, 50.290 ops/sec
ECDSA    256 verify         50 ops took 1.025 sec, avg 20.500 ms, 48.780 ops/sec
Benchmark complete

Benchmark Test Completed

```

Please check if my assumptions are correct.
Regards
Sylwester

dgarske, 2020-06-18:
Hi SylwBar,

Thanks for these excellent build error reports. A quick glance looks like your analysis and fixes are correct. I'll make sure these get fixed and a PR put up with a link here.

David Garske, wolfSSL

SylwBar, 2020-06-19:
I just run CryptoCell tests and confirm the changes, thank you for the quick fix.
Regards
Sylwester

PR #3060 (2020-06-22): Fixes for CryptoCell
* Fix for signature wrapper signing to allow larger signing input buffer.
* Fix for bad cryptocell ECC make key result check (-9628).
* Cleanup of some duplicate code in signature wrapper. 

Tested on nRF52840-DK (PCA10056).

Fixes #3059. Thanks Sylwester.
commit 6b1a6309 (2020-06-22): Fixes for CryptoCell. Fix for signature wrapper signing to allow larger signing input buffer. Cleanup of some duplicate code. Fix for bad cryptocell ECC make key result check (-9628). Fixes #3059. Thanks Sylwester.
closed 2020-06-22 by toddouska