wolfSSL/wolfssl #3070: CRL Lookup after failing OCSP

argold97, 2020-06-23:
In `internal.c` `ProcessPeerCerts()` line `10913`, the code seems to be structured in such a way if you have both OCSP and CRL enabled Wolf will first attempt to use OCSP to verify a peer cert, and fallback to CRL if OCSP was unsuccessful. However the code never actually allows CRL checking to happen after OCSP:

```
                        if (ssl->ctx->cm->ocspEnabled &&
                                            ssl->ctx->cm->ocspCheckAll) {
                            WOLFSSL_MSG("Doing Non Leaf OCSP check");
                            ret = CheckCertOCSP_ex(ssl->ctx->cm->ocsp,
                                                    args->dCert, NULL, ssl);
                        #ifdef WOLFSSL_NONBLOCK_OCSP
                            if (ret == OCSP_WANT_READ) {
                                args->lastErr = ret;
                                goto exit_ppc;
                            }
                        #endif
                            doCrlLookup = (ret == OCSP_CERT_UNKNOWN);
                            if (ret != 0) {
                                doCrlLookup = 0;
                                WOLFSSL_MSG("\tOCSP Lookup not ok");
                            }
                        }
                #endif /* HAVE_OCSP */

                #ifdef HAVE_CRL
                        if (ret == 0 && doCrlLookup &&
                                    ssl->ctx->cm->crlEnabled &&
                                                ssl->ctx->cm->crlCheckAll) {
```

It seems like after OCSP fails it tries to do a CRL check by doing:
`doCrlLookup = (ret == OCSP_CERT_UNKNOWN)`
However this is immediately overridden by the next two lines:
```
                           if (ret != 0) {
                                doCrlLookup = 0;
                                WOLFSSL_MSG("\tOCSP Lookup not ok");
                            }
```
Furthermore the CRL lookup cannot happen since `ret` will not be zero when reaching this line:
```
                #ifdef HAVE_CRL
                        if (ret == 0 && doCrlLookup &&
                                    ssl->ctx->cm->crlEnabled &&
                                                ssl->ctx->cm->crlCheckAll) {
```

So I can't tell from the code whether doing CRL after failing to do OCSP is the intended use case. If not, is there a way to do this?

dgarske, 2020-06-23:
Hi @argold97 ,  thanks for the bug report. We will take a look this and fix. Is there a certificate or website you are using to test against that you can share?

Thanks,
David Garske, wolfSSL

argold97, 2020-06-23:
Sorry, the certificates I'm using are private. But it should be easy to make a test certificate to reproduce this issue by creating one with a fake OCSP responder URL. After it fails to connect to this fake responder it should try to do CRL.

kaleb-himes, 2020-06-30:
@ejohnstown can you assign this maintenance item please. Thanks!

- KH

ejohnstown, 2022-11-24:
I'm fixing this by restoring the original behavior from 2014, while keeping the existing behavior updates. The original way of doing OCSP then resorting to CRL made the most sense.

closed 2022-11-24 by ejohnstown