wolfSSL/wolfssl #3093: [Question] Has wolfSSL completed support for the Windows kernel?

Jeremy203, 2020-07-01:
Hello!

I'm a Windows kernel developer and I was recently looking for a way to use SSL in the kernel to communicate securely with the server.

I did a Google search and found an article about wolfSSL from 10 years ago. ([Link](https://www.wolfssl.com/windows-kernel-mode-compatibility-for-wolfssl-2/))

I would like to know if wolfSSL now has completed support for the Windows kernel and some sample code?

If not, can I easily port them to the kernel using the latest version of the wolfSSL code?
(I've done some simple socket operations in the kernel, like connect, send, receive, etc.)

Thank you!

kaleb-himes, 2020-07-01:
Hi Jeremy203,

Thank you for reaching out. wolfSSL has not done any examples ourselves but we have many customers using wolfSSL in the kernel space on both windows and linux. We have had very few questions about it other than "is it possible" to which the answer is always "yes!" so while we haven't ported it and can not personally attest to the porting simplicity, given the relatively few questions and large volume of use-cases we expect it is not overly difficult to do.

Hope this helps, let us know if you have any followup questions.

- KH

Jeremy203, 2020-07-02:
Hello, kaleb-himes!

Thanks for the reply, but I noticed that the wolfSSL license is a bit unacceptable to me, so I sadly have to prioritize other SSL libraries.

Thank you! Have a good day!

closed 2020-07-02 by Jeremy203