wolfSSL/wolfssl #3096: CryptoCell port: wc_Sha256FinalRaw() function is not provided.

SylwBar, 2020-07-01:
Hi.
 I got such compiler error during compiling example DTLS application with WOLFSSL_CRYPTOCELL flag enabled:

```
In function `Hmac_HashFinalRaw':
...wolfssl/src/tls.c:773: undefined reference to `wc_Sha256FinalRaw'.
```
I think this function should be added to: cryptoCellHash.c.
There is similar wc_Sha256Final() defined already in this file.

Regards
Sylwester

tmael, 2020-07-01:
Hi Sylwester,

Which configuration options are you using? 
Cryptocell was tested with wolfCrypt, not with TLS/DTLS.
Could you please try making the following changes in tls.c and let me know if it works? 
```
#if !defined(NO_SHA256) && defined(WOLFSSL_CRYPTOCELL)
            ret = wc_Sha256Final(&hmac->hash.sha256, hash);
#else
            ret = wc_Sha256FinalRaw(&hmac->hash.sha256, hash);
#endif 
```

Thanks,
Tesfa

SylwBar, 2020-07-02:
Hi Tesfa.
I applied your changes, and my project compiled, however with plenty of warnings in ecc.c.
Those warnings are visible only with WOLFSSL_CRYPTOCELL defined.

Some background:
Without WOLFSSL_CRYPTOCELL defined my project is able to perform complete DTLS 1.2 handshake and send some data.
It is acting as DTLS client connecting to DTLS server on Linux over IPv6 BLE link.
Target board is NRF52840DK.
With WOLFSSL_CRYPTOCELL defined completing handshake is not possible even with your SHA256 related changes.
All settings are identical except mentioned define.

I'm attaching several files:
* user_settings.h - are almost all settings, additionally there are some more in makefile:
-DBOARD_PCA10056
-DNRF52840_XXAA
-DWOLFSSL_USER_SETTINGS
-DDEBUG_WOLFSSL

* DTLS_handshake_SHA256_SW.txt - WolfSSL log taken from NRF board during successful handshake - no WOLFSSL_CRYPTOCELL defined - pure software,
* DTLS_handshake_SHA256_SW.pcapng - wireshark log,
* DTLS_handshake_SHA256_CCell.txt - log taken during DTLS handshake but with WOLFSSL_CRYPTOCELL defined,
* DTLS_handshake_SHA256_CCell.pcapng - wireshark log.

Cryptocell port exposes problem with signature verification:
00> SSL[1]: CRYS_ECDSA_Verify failed
00> SSL[3]: wolfSSL Leaving EccVerify, return -330 

I also check if the problem is related to SHA256 or something else. I enabled SHA384 (user_settings.h:384) so handshake process switched to SHA384, but results are the same: error -330.
SHA384 is not supported by CryptoCell, it is using SW implementation.

This confirms that DTLS with CC310 enabled needs more attention, not only SHA256 fix.
But you already mentioned that TLS/DTLS was not tested with CC310.
I'll be glad to see complete CC310 port implementation :)

[logs.zip](https://github.com/wolfSSL/wolfssl/files/4865197/logs.zip)

tmael, 2020-07-02:
Hi Sylwester,

Thanks for sending more info. 

The supported and tested features for CC310 are documented at 
https://github.com/wolfSSL/wolfssl/blob/master/IDE/CRYPTOCELL/README.md

We'd be more than happy to add the TLS/DTLS port into our growing backlog. 

If there is a timeline associated with having this port available please let me know so we can expedite the request. 

Thanks,
Tesfa

SylwBar, 2020-07-07:
Hi.
I'm evaluating WolfSSL/DTLS features - lack of CC310 port is not blocking me.
I wish I could evaluate performance improvements offered by CC310 port, but it is not priority at the moment.
For sure I'd like to see complete CC310 port completed in future.

Please handle the issue according to your policy.

Regards
Sylwester

tmael, 2020-07-10:
Hi Sylwester,

Thank you so much for officially registering your feature request with wolfSSL. Please find below a summary of the requested feature:

Requested Feature: TLS/DTLS with Cryptocell
Requestor Name: Sylwester Barański 
Requestor Email: <Requestor Email>
Requested Source: WolfSSL GitHub issue #3096
Requested Date: July 7, 2020
Implementation Date: Unfunded effort, no timeline

We try to work on requested features whenever we get free engineering cycles unless a customer decides to back the effort for a finite implementation timeline.  If at any time a feature becomes a progress blocker for a project it can be accelerated by contacting  support@wolfssl.com.

Please let me know if there is anything else we can assist with at this time.

Thanks,
Tesfa

tmael, 2020-12-15:
Closing since feature request has already been added in our backlog. If you need an update you can always contact support@wolfssl.com.

closed 2020-12-15 by tmael