wolfSSL/wolfssl #3101: how to set --enable-all option in MSVC or user_settings.h ?

myvcgit, 2020-07-04:
i use wolfssl64.sln compile wolfssl static lib with MSVC,after compile libcurl with wolfssl.lib.
when i use this libcurl and set curl_easy_setopt(curl, CURLOPT_HTTP_VERSION, CURL_HTTP_VERSION_2_0)
but  can't use http/2 just use http/1.1,the log:

libcurl/7.71.2-DEV wolfSSL/4.4.1 zlib/1.2.11 brotli/1.0.7 nghttp2/1.41.90
*   Trying 151.101.174.49:443...
* Connected to curl.haxx.se (151.101.174.49) port 443 (#0)
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
> GET /libcurl/c/http2-download.html HTTP/1.1
Host: curl.haxx.se
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Connection: keep-alive
< Content-Length: 21602
....

i think maybe the compile settings wrong,i found --enable-all option but i don't know how to set that in MSVC or user_settings.h.

i use vs2019 16.6.0 and win 10 x64

my user_settings.h setting:

#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_

#define WC_RSA_BLINDING

#define OPENSSL_ALL
#define OPENSSL_EXTRA

#define WOLFSSL_RIPEMD
#define NO_PSK
#define HAVE_EXTENDED_MASTER
#define HAVE_TLS_EXTENSIONS
#define HAVE_ALPN
#define WOLFSSL_ALWAYS_KEEP_SNI
#define WOLFSSL_ALLOW_TLSV10
#define WOLFSSL_TLS13
#define HAVE_AES_ECB
#define HAVE_AES_KEYWRAP
#define HAVE_AESGCM
#define HAVE_AESCCM
#define WOLFSSL_AES_CFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_XTS
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define HAVE_ECC
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_KOBLITZ
#define HAVE_ECC_SECPR2
#define HAVE_ECC_BRAINPOOL
#define HAVE_ECC_SECPR3
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define WOLFSSL_CUSTOM_CURVES
#define HAVE_SUPPORTED_CURVES
#define HAVE_SECURE_RENEGOTIATION
#define WC_NO_HARDEN
#define WOLFSSL_KEY_GEN
#define HAVE_COMP_KEY
#define WOLFSSL_DES_ECB
#define HAVE_FFDHE_2048
#define WC_RSA_PSS

#define HAVE_PKCS7
#define HAVE_HKDF
#define HAVE_X963_KDF
#define WOLFSSL_CERT_EXT
#define WOLFSSL_CERT_GEN
#define WOLFSSL_CERT_REQ
#define WOLFSSL_DER_LOAD
#define WOLFSSL_DTLS
#define HAVE_CAMELLIA
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_ONE_TIME_AUTH
#define HAVE_CRL
#define HAVE_CRL_IO
#define HAVE_OCSP
#define WOLFSSL_SNIFFER
#define WOLFSSL_SNIFFER_WATCH
#define WOLFSSL_SNIFFER_STATS
#define HAVE_CURVE25519

/* Optional Performance Speedups /
#if 1
/ AESNI on x64 */
#if(defined(_WIN64) || defined(USE_ARCH_X86_64) || defined(USE_ARCH_AMD64))
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#endif

/* Single Precision Support for RSA/DH 1024/2048/3072 and ECC P-256 */
#define WOLFSSL_SP
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_RSA

#if _DEBUG
#define DEBUG_WOLFSSL
#endif

#endif /* _WIN_USER_SETTINGS_H_ */

kaleb-himes, 2020-07-06:
Can you give these a try, this is equivalent to the configure --enable-all:

```
#ifndef WOLFSSL_USER_SETTINGS_H
#define WOLFSSL_USER_SETTINGS_H

#ifdef __cplusplus
extern "C" {
#endif

#undef  WOLFSSL_PUBLIC_MP
#define WOLFSSL_PUBLIC_MP

#undef  WOLFSSL_TLS13
#define WOLFSSL_TLS13

#undef  HAVE_TLS_EXTENSIONS
#define HAVE_TLS_EXTENSIONS

#undef  HAVE_SUPPORTED_CURVES
#define HAVE_SUPPORTED_CURVES

#undef  HAVE_FFDHE_2048
#define HAVE_FFDHE_2048

#undef  WOLFSSL_SEP
#define WOLFSSL_SEP

#undef  KEEP_PEER_CERT
#define KEEP_PEER_CERT

#undef  WOLFSSL_VERIFY_CB_ALL_CERTS
#define WOLFSSL_VERIFY_CB_ALL_CERTS

#undef  WOLFSSL_EXTRA_ALERTS
#define WOLFSSL_EXTRA_ALERTS

#undef  OPENSSL_EXTRA
#define OPENSSL_EXTRA

#undef  WOLFSSL_ALWAYS_VERIFY_CB
#define WOLFSSL_ALWAYS_VERIFY_CB

#undef  OPENSSL_ALL
#define OPENSSL_ALL

#undef  WOLFSSL_EITHER_SIDE
#define WOLFSSL_EITHER_SIDE

#undef  WC_RSA_NO_PADDING
#define WC_RSA_NO_PADDING

#undef  WC_RSA_PSS
#define WC_RSA_PSS

#undef  WOLFSSL_DTLS
#define WOLFSSL_DTLS

#ifndef WOLFSSL_OPTIONS_IGNORE_SYS
#undef  _POSIX_THREADS
#define _POSIX_THREADS
#endif

#undef  WOLFSSL_MULTI_ATTRIB
#define WOLFSSL_MULTI_ATTRIB

#undef  HAVE_THREAD_LS
#define HAVE_THREAD_LS

#undef  HAVE_AES_DECRYPT
[... 487 lines cut ...]

#undef  NO_OLD_SSL_NAMES
#define NO_OLD_SSL_NAMES

#undef  NO_OLD_SHA_NAMES
#define NO_OLD_SHA_NAMES

#undef  WOLFSSL_HASH_FLAGS
#define WOLFSSL_HASH_FLAGS

#undef  WOLFSSL_HAVE_WOLFSCEP
#define WOLFSSL_HAVE_WOLFSCEP

#undef  HAVE_PKCS7
#define HAVE_PKCS7

#undef  HAVE___UINT128_T
#define HAVE___UINT128_T 1

#ifdef __cplusplus
}
#endif

#endif /* WOLFSSL_USER_SETTINGS_H */
```

myvcgit, 2020-07-07:
@kaleb-himes  Thank you

i add to ../ide/win/user_settings.h,but have some problem，need comment some define

//wolfssl\wolfssl\wolfcrypt\fe_448.h(51,34): error C2061: syntax error: identifier '__attribute__'
// #undef  HAVE_CURVE448
// #define HAVE_CURVE448

//wolfssl\src\crl.c(43,1): fatal error C1189: #error:  "CRL monitor only currently supported on linux or mach"
// #undef  HAVE_CRL_MONITOR
// #define HAVE_CRL_MONITOR

// wolfssl\wolfcrypt\src\ge_operations.c(683,13): error C2065: '__int128_t': undeclared identifier
// #undef  HAVE___UINT128_T
// #define HAVE___UINT128_T 1

this my user_settings.h file

#ifndef WOLFSSL_USER_SETTINGS_H
#define WOLFSSL_USER_SETTINGS_H

#ifdef __cplusplus
extern "C" {
#endif

#undef  WOLFSSL_PUBLIC_MP
#define WOLFSSL_PUBLIC_MP

#undef  WOLFSSL_TLS13
#define WOLFSSL_TLS13

#undef  HAVE_TLS_EXTENSIONS
#define HAVE_TLS_EXTENSIONS

#undef  HAVE_SUPPORTED_CURVES
#define HAVE_SUPPORTED_CURVES

#undef  HAVE_FFDHE_2048
#define HAVE_FFDHE_2048

#undef  WOLFSSL_SEP
#define WOLFSSL_SEP

#undef  KEEP_PEER_CERT
#define KEEP_PEER_CERT

#undef  WOLFSSL_VERIFY_CB_ALL_CERTS
#define WOLFSSL_VERIFY_CB_ALL_CERTS

#undef  WOLFSSL_EXTRA_ALERTS
#define WOLFSSL_EXTRA_ALERTS

#undef  OPENSSL_EXTRA
#define OPENSSL_EXTRA

#undef  WOLFSSL_ALWAYS_VERIFY_CB
#define WOLFSSL_ALWAYS_VERIFY_CB

#undef  OPENSSL_ALL
#define OPENSSL_ALL

#undef  WOLFSSL_EITHER_SIDE
#define WOLFSSL_EITHER_SIDE

#undef  WC_RSA_NO_PADDING
#define WC_RSA_NO_PADDING

#undef  WC_RSA_PSS
#define WC_RSA_PSS

[... 504 lines cut ...]

#undef  NO_OLD_SSL_NAMES
#define NO_OLD_SSL_NAMES

#undef  NO_OLD_SHA_NAMES
#define NO_OLD_SHA_NAMES

#undef  WOLFSSL_HASH_FLAGS
#define WOLFSSL_HASH_FLAGS

#undef  WOLFSSL_HAVE_WOLFSCEP
#define WOLFSSL_HAVE_WOLFSCEP

#undef  HAVE_PKCS7
#define HAVE_PKCS7

// wolfssl\wolfcrypt\src\ge_operations.c(683,13): error C2065: '__int128_t': undeclared identifier
// #undef  HAVE___UINT128_T
// #define HAVE___UINT128_T 1

#ifdef __cplusplus
}
#endif

#endif /* WOLFSSL_USER_SETTINGS_H */

dgarske, 2020-07-09:
FYI: There is also a PR for `user_settings.h` templates here:
https://github.com/wolfSSL/wolfssl/pull/2882

One is an equivalent to `--enable-all` with formatting and comments:
https://github.com/wolfSSL/wolfssl/pull/2882/files#diff-521976eac614a92de7b44fa4614016cd

Thanks,
David Garske, wolfSSL

myvcgit, 2020-07-11:
Thanks all

closed 2020-07-11 by myvcgit