wolfSSL/wolfssl #3117: Possible use word32 before define when use in Contiki-NG

kam193, 2020-07-08:
When use on Contiki-NG, `word32` is used before defined because of use in `settings.h`.

`word32` is defined in `types.h`
https://github.com/wolfSSL/wolfssl/blob/6196698d8b3d8c2338a193bcb73ead7550befe96/wolfssl/wolfcrypt/types.h#L29
https://github.com/wolfSSL/wolfssl/blob/6196698d8b3d8c2338a193bcb73ead7550befe96/wolfssl/wolfcrypt/types.h#L68

But it first includes `settings.h` which contains following define:

https://github.com/wolfSSL/wolfssl/blob/80246dfbc370840e13a7f177803c958a7dbc7218/wolfssl/wolfcrypt/settings.h#L316-L330

kam193, 2020-07-08:
Going deeper:
* in `internal.c` there is a place for defining `LowResTimer`, but without options for `WOLFSSL_CONTIKI`. So it's not only using `word32` before defining, but also redefining of function
* using `random_rand` requires including `lib/random.h` (from Contiki-NG)

danielinux, 2020-07-09:
Hello @kam193 - thanks for reporting this.

There is a patch we maintain on top of [contiki-os/contiki](github.com/contiki-os/contiki), but I did not look into a port for contiki-ng yet.

On contiki-os we created this glue module that can be used to initialize and connect TLS and DTLS sockets on top of Contiki's uIP sockets, which you can find [here](https://github.com/danielinux/contiki/tree/master/apps/wolfssl).
The `WOLFSSL_CONTIKI` define is intended for use with that module, but I am not sure that it could be adapted "as is" in Contiki-NG as well. 

Is there a way to include third-party modules like in the original contiki-os? What would be the best way to propose a wolfSSL module for inclusion there? I see that the directory structure is different than the original contiki-os, and there is no apps/ directory. 
I have quickly looked into the contiki-ng documentation but I could not find any indication on how to integrate third party libraries in the repository. 

Would you mind sharing with us more details about your current configuration and environment setup? We'll be happy to help defining a generic platform for TLS/DTLS sockets, and make the required configuration options available in the library for proper integration in the OS.

Thanks,

-- 
Daniele

kam193, 2020-07-09:
Hello,

I'm currently during investigation how to integrate wolfSSL and Contiki-NG in the clean way in my project (but I probably will need only a wolfcrypt, so maybe my integration will contain only part). I'm basing e.g. on PR with your module (https://github.com/contiki-os/contiki/pull/2500). 

Unfortunately I cannot answer your questions with 100% sure. Currently, to compile Contiki-NG with wolfSSL I use extending `CONTIKI_SOURCEFILES` and `CFLAGS` in Makefile (very similar to your PR; the build system is partially described [here](https://github.com/contiki-ng/contiki-ng/wiki/The-Contiki%E2%80%90NG-build-system)). When I find the best way to integrate wolfSSL, I will back here. 

The main problem I reported - redefining `LowResTimer` - looks like not Contiki nor Conitki-NG specific, but rather like forgotten when other definitions was moved into `internal.c`.

For now, I see a few other small incompatibilities with Contiki-NG caused by missed including headers. I think I'll prepare a PR with required changes after successfully integration.

danielinux, 2020-07-09:
Thanks for the clarification! We'll be looking forward to receiving your proposed changes.

Best Regards,

--
Daniele

danielinux, 2021-06-15:
Hi @kam193 
I'm closing this issue for now. Feel free to re-open if you find a configuration that works with contiki-ng code, or if you want to propose changes.

Best Regards,

-- 
Daniele

closed 2021-06-15 by danielinux