wolfSSL/wolfssl #3136: issue related to CUSTOM_RAND_GENERATE_BLOCK

ghs-sec, 2020-07-16:
Hi wolfssl team,
when using `CUSTOM_RAND_GENERATE_BLOCK`, the following in `/wolcrypt/src/random.c` is enabled:
```
#ifdef CUSTOM_RAND_GENERATE_BLOCK
    XMEMSET(output, 0, sz);
    ret = CUSTOM_RAND_GENERATE_BLOCK(output, sz);
```
Later it is suggested:
```
/* #define CUSTOM_RAND_GENERATE_BLOCK myRngFunc
     * extern int myRngFunc(byte* output, word32 sz);
     */
```

**Problem**:
`#define CUSTOM_RAND_GENERATE_BLOCK myRngFunc` causes compiler to complain redefinition of `CUSTOM_RAND_GENERATE_BLOCK`.

**suggestion**:
It would be good to leave: `ret = myRngFunc(output, sz);` and put a forward declaration of  `myRngFunc`. 
This leaves the implementer to supply its own `myRngFunc`.

**Testing environment**:
gcc-9.3 and ubuntu 20.04

dgarske, 2020-07-16:
Hi @ghs-sec , 

The `myRngFunc` is just an example. You can name your function anything. 

The example for `CUSTOM_RAND_GENERATE_BLOCK` down around line 2375 is commented out and should not cause a redefinition error.

You can find an example of `CUSTOM_RAND_GENERATE_BLOCK` here:
https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h#L482

Thanks,
David Garske, wolfSSL

ghs-sec, 2020-07-17:
Thanks a lot for your help!

closed 2020-07-17 by ghs-sec