wolfSSL/wolfssl #3138: unit tests fail with SCTP enabled

tycho, 2020-07-16:
After applying the compile fix in #3137 and building with `--enable-sctp`, it seems the SCTP unit tests fail:
```
trying server command line[1724]: SuiteTest -G -v 2 -l DHE-RSA-CHACHA20-POLY1305 -2 -p 0 -H defCipherList 
Using default cipher list for testing
trying client command line[1725]: SuiteTest -G -v 2 -l DHE-RSA-CHACHA20-POLY1305 -2 -p 43952 
SSL_accept error -501, can't match cipher suite
wolfSSL error: SSL_accept failed
wolfSSL_connect error -313, received alert fatal error
```
I can reproduce the issue by manually running a pair of `examples/server/server` and `examples/client/client` with arguments matching the test. They do seem to connect, but somehow the negotiation fails.

(And I've verified SCTP is working fine at the kernel level by just doing some `socat` conversations.)

ejohnstown, 2020-07-20:
While the server is getting the cipher suite DHE-RSA-CHACHA20-POLY1305, the -H option makes the server ignore that request. It is running with a default cipher suite list, and the client requesting that one causes the fail. If you delete the -H option, it'll work.

tycho, 2020-07-20:
Then maybe this shouldn't be a test failure if it's expected?

ejohnstown, 2020-07-20:
It's a couple things. Mostly it was premature of me to post that reply. I have a fix I'm testing at the moment. It's a configuration issue.

tycho, 2020-07-20:
Gotcha. Thanks for looking!

ejohnstown, 2020-07-20:
Please see PR #3150 for a fix.

ejohnstown, 2020-07-22:
And it has been merged.

closed 2020-07-22 by ejohnstown