wolfSSL/wolfssl #3143: Compilation error for optee ARM architecture

geneva1, 2020-07-17:
I am trying to compile wolfssl for optee ARM architecture.

I found [a wolfssl developer write-up](https://translate.google.com/translate?sl=auto&tl=en&u=https%3A%2F%2Fm.blog.naver.com%2FPostView.nhn%3FblogId%3Dhw5773%26logNo%3D221319208008%26proxyReferer%3Dhttps%3A%252F%252Fwww.google.com%252F) about it and followed the instructions there.

However, I have this mfloat-abi error   [fatal error: gnu/stubs-soft.h: No such file or directory](https://stackoverflow.com/questions/49139125/fatal-error-gnu-stubs-soft-h-no-such-file-or-directory)

Why ?

    make[1]: Entering directory '/home/rog/Documents/optee-qemu/optee_os'
      CHK     out/arm/conf.mk
      CHK     out/arm/include/generated/conf.h
      CC      out/arm/core/pta/secstor_ta_mgmt.o
    In file included from out/arm/core/include/features.h:452,
                     from out/arm/core/include/pthread.h:21,
                     from out/arm/core/include/wolfssl/wolfcrypt/wc_port.h:160,
                     from out/arm/core/include/wolfssl/wolfcrypt/types.h:30,
                     from out/arm/core/include/wolfssl/wolfcrypt/logging.h:33,
                     from out/arm/core/include/wolfssl/ssl.h:35,
                     from core/pta/secstor_ta_mgmt.c:22:
    out/arm/core/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-soft.h: No such file or directory
     # include <gnu/stubs-soft.h>
               ^~~~~~~~~~~~~~~~~~
    compilation terminated.
    make[1]: *** [mk/compile.mk:148: out/arm/core/pta/secstor_ta_mgmt.o] Error 1
    make[1]: Leaving directory '/home/rog/Documents/optee-qemu/optee_os'
    make: *** [common.mk:378: optee-os-common] Error 2
    rog@rog-GL502VMK:~/Documents/optee-qemu/build$ cd ../
    rog@rog-GL502VMK:~/Documents/optee-qemu$
    rog@rog-GL502VMK:~/Documents/optee-qemu$ grep -rinIe "mfloat-abi"
    u-boot/arch/nds32/config.mk:18:PLATFORM_RELFLAGS += -fno-common -mrelax -mno-ext-fpu-dp -mfloat-abi=soft
    u-boot/arch/arm/lib/asm-offsets.s:26:@ -mno-unaligned-access -mfloat-abi=soft -mtune=generic-armv7-a -mfpu=neon
    u-boot/lib/asm-offsets.s:26:@ -mno-unaligned-access -mfloat-abi=soft -mtune=generic-armv7-a -mfpu=neon
    out-br/build/libopenssl-1.1.1a/Configurations/15-android.conf:193:        # -march and/or -mfloat-abi flags. NDK defaults to armv5te.
    out-br/build/host-python-2.7.15/Modules/_ctypes/libffi/ChangeLog:3552:    -mfloat-abi defaults to 'softfp' instead of 'soft'. To work around this
    out-br/build/host-python-2.7.15/Modules/_ctypes/libffi/ChangeLog.libffi-3.1:1665: and compiler options include -mfloat-abi=hard.
    out-br/build/host-python-2.7.15/Modules/_ctypes/libffi/ChangeLog.libffi-3.1:2130: for -mfloat-abi=softfp to work.
    out-br/build/host-gmp-6.1.2/configure:4269:        gcc_cflags_fpmode="-mfloat-abi=softfp" ;;
    out-br/build/host-gmp-6.1.2/configure:4271:        gcc_cflags_fpmode="-mfloat-abi=hard" ;;
    out-br/build/host-gmp-6.1.2/configure.ac:611:        gcc_cflags_fpmode="-mfloat-abi=softfp" ;;
    out-br/build/host-gmp-6.1.2/configure.ac:613:        gcc_cflags_fpmode="-mfloat-abi=hard" ;;
    ^C
    rog@rog-GL502VMK:~/Documents/optee-qemu$

dgarske, 2020-07-20:
Hi @geneva1,

Thanks for sharing that blog post for OPTEE. I had not seen that effort. We've done some integration and testing with OPTEE officially, but that work is still in progress here https://github.com/dgarske/optee_os/tree/wolfssl.

These errors appear to be cross-compilation errors between target and host with pthread. Possibly related to the sysroot. It looks like Tongue's Blog disables the pthread support. That is done using `CFLAGS="-DSINGLE_THREADED"` or `--enable-singlethreaded`.

It would be great to hear more about your efforts. Feel free to send us a direct email at support@wolfssl.com to enlist our direct support.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-21:
@dgarske can you show the steps to compile https://github.com/dgarske/optee_os/tree/wolfssl ?

dgarske, 2020-07-21:
Hi @geneva1 , 

My work with OPTEE-OS was part of an STM Telemaco 3P effort, which used Yocto linux. So the actual building was being done with Yocto meta-recipes.

https://github.com/dgarske/optee_os/tree/wolfssl/lib/libwolfssl
https://github.com/dgarske/meta-wolfssl/tree/stm32_ehsm
https://github.com/dgarske/wolfssl/tree/stm32_telemaco3p

Note: This work is only about 75% complete.

For reference this issue is also being tracked here:
https://github.com/OP-TEE/optee_os/issues/3980

It looks like your current error is due to your OPTEE-OS code not including "wolfssl/options.h" prior to any other wolfSSL headers to pull in the build configuration used with ./configure. The `./configure` generates `wolfssl/options.h`, which has all the build defines.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-21:
> not including "wolfssl/options.h" prior to any other wolfSSL headers to pull in the build configuration used with ./configure. 

I have already included `wolfssl/options.h` prior to any other wolfSSL headers inside [secstor_ta_mgmt.c](https://github.com/wolfSSL/wolfssl/files/4954771/secstor_ta_mgmt.txt)
I do not understand what you are trying to describe.

```
make[1]: Entering directory '/home/rog/Documents/optee-qemu/optee_os'
  CHK     out/arm/conf.mk
  CHK     out/arm/include/generated/conf.h
  CC      out/arm/core/pta/secstor_ta_mgmt.o
In file included from out/arm/core/include/features.h:452,
                 from out/arm/core/include/pthread.h:21,
                 from out/arm/core/include/wolfssl/wolfcrypt/wc_port.h:160,
                 from out/arm/core/include/wolfssl/wolfcrypt/types.h:30,
                 from out/arm/core/include/wolfssl/wolfcrypt/logging.h:33,
                 from out/arm/core/include/wolfssl/ssl.h:35,
                 from core/pta/secstor_ta_mgmt.c:7:
out/arm/core/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-soft.h: No such file or directory
 # include <gnu/stubs-soft.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [mk/compile.mk:148: out/arm/core/pta/secstor_ta_mgmt.o] Error 1
make[1]: Leaving directory '/home/rog/Documents/optee-qemu/optee_os'
make: *** [common.mk:378: optee-os-common] Error 2
rog@rog-GL502VMK:~/Documents/optee-qemu/build$ 
```

```
// SPDX-License-Identifier: BSD-2-Clause
/*
 * Copyright (c) 2017, Linaro Limited
 */

#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <kernel/pseudo_ta.h>
#include <tee/tadb.h>
#include <pta_secstor_ta_mgmt.h>
#include <signed_hdr.h>
#include <string_ext.h>
#include <string.h>
#include <tee_api_types.h>
#include <crypto/crypto.h>
#include <tee/uuid.h>
#include <types_ext.h>
#include <utee_defines.h>

#include <tee_internal_api.h>
#include <tee_internal_api_extensions.h>
#include <tee_tcpsocket.h>
```

dgarske, 2020-07-21:
Hi @geneva1 ,

Again the issue here is cross-compiling with pthreads. If your `wolfssl/options.h` contains `SINGLE_THREADED` then the code will not include `pthread.h` and you should not have a build error here.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-21:
@dgarske I have added `SINGLE_THREADED` to [wolfssl/options.h](https://github.com/wolfSSL/wolfssl/files/4955105/options.txt) , but I still have the same build error...

```
#ifndef SINGLE_THREADED
#define SINGLE_THREADED
```

dgarske, 2020-07-21:
Hi @geneva1 , 

You should double check its including the correct `wolfssl/options.h`. You may need to adjust your include order. Sometimes I find it helpful to put an `#error test` to force a build error to test if its including the right file. My guess is that it's including your `/usr/local/include/wolfssl/options.h`. If you have `SINGLE_THREADED` defined `pthread.h` will not be included.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-22:
I have already added `SINGLE_THREADED` to all four `wolfssl/options.h` files, but same build error

```
rog@rog-GL502VMK:~/Documents/optee-qemu$ find ./ -name "options.h"
./u-boot/include/config/board/specific/options.h
./u-boot/include/config/sys/extra/options.h
./out-br/build/buildroot-config/br2/tar/options.h
./out-br/build/buildroot-config/br2/target/generic/getty/options.h
./out-br/build/buildroot-config/br2/target/rootfs/tar/options.h
./out-br/build/busybox-1.29.3/include/config/ifupdown/udhcpc/cmd/options.h
./out-br/build/busybox-1.29.3/include/config/feature/cmdline/module/options.h
./out-br/build/busybox-1.29.3/include/config/feature/tar/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/nslookup/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/ftpgetput/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/chown/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/run/parts/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/gunzip/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/start/stop/daemon/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/install/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/cp/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/setconsole/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/diff/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/gzip/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/ipcalc/long/options.h
./out-br/build/busybox-1.29.3/include/config/feature/wget/long/options.h
./toolchains/aarch32/lib/gcc/arm-linux-gnueabihf/8.3.0/plugin/include/options.h
./toolchains/aarch64/lib/gcc/aarch64-linux-gnu/8.3.0/plugin/include/options.h
./linux/include/config/jffs2/compression/options.h
./linux/include/config/media/common/options.h
./optee_os/out/arm/core/include/cyassl/options.h
./optee_os/out/arm/core/include/wolfssl/options.h
./optee_os/out/arm/include/wolfssl/options.h
./optee_os/out/arm/export-ta_arm32/include/cyassl/options.h
./optee_os/out/arm/export-ta_arm32/include/wolfssl/options.h
./optee_os/lib/libwolfssl/cyassl/options.h
./optee_os/lib/libwolfssl/wolfssl/options.h
rog@rog-GL502VMK:~/Documents/optee-qemu$
```

geneva1, 2020-07-22:
As for `#error test`, see the following :

```
rog@rog-GL502VMK:~/Documents/optee-qemu/build$ make
make -C /home/rog/Documents/optee-qemu/build/../optee_os O=out/arm PLATFORM=vexpress-qemu_virt CROSS_COMPILE="/usr/bin/ccache /home/rog/Documents/optee-qemu/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CROSS_COMPILE_core="/usr/bin/ccache /home/rog/Documents/optee-qemu/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CROSS_COMPILE_ta_arm64="/usr/bin/ccache /home/rog/Documents/optee-qemu/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_ta_arm32="/usr/bin/ccache /home/rog/Documents/optee-qemu/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" CFG_TEE_CORE_LOG_LEVEL=3 DEBUG=1 CFG_TEE_BENCHMARK=n
make[1]: Entering directory '/home/rog/Documents/optee-qemu/optee_os'
  CHK     out/arm/conf.mk
  CHK     out/arm/include/generated/conf.h
  CC      out/arm/core/pta/secstor_ta_mgmt.o
In file included from core/pta/secstor_ta_mgmt.c:6:
out/arm/core/include/wolfssl/options.h:9:2: error: #error test
 #error test
  ^~~~~
out/arm/core/include/wolfssl/options.h:10: error: unterminated #ifndef
 #ifndef WOLFSSL_OPTIONS_H
 
In file included from out/arm/core/include/features.h:452,
                 from out/arm/core/include/dirent.h:25,
                 from out/arm/core/include/wolfssl/wolfcrypt/wc_port.h:441,
                 from out/arm/core/include/wolfssl/wolfcrypt/types.h:30,
                 from out/arm/core/include/wolfssl/wolfcrypt/logging.h:33,
                 from out/arm/core/include/wolfssl/ssl.h:35,
                 from core/pta/secstor_ta_mgmt.c:7:
out/arm/core/include/gnu/stubs.h:7:11: fatal error: gnu/stubs-soft.h: No such file or directory
 # include <gnu/stubs-soft.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [mk/compile.mk:148: out/arm/core/pta/secstor_ta_mgmt.o] Error 1
make[1]: Leaving directory '/home/rog/Documents/optee-qemu/optee_os'
make: *** [common.mk:378: optee-os-common] Error 2
rog@rog-GL502VMK:~/Documents/optee-qemu/build$ 
```

dgarske, 2020-07-22:
Hi @geneva1 ,

That shows you which options.h is being used. There really should only be one. You got passed the pthread.h and now the issue is with dirent.h. Either you need to resolve the cross-compiler target / host issue to resolve this or disable the file system support in wolfSSL using "--disable-file system" or "#define NO_FILESYSTEM".

I think it would be useful to setup a call to help support you and talk about your project. Please email support@wolfssl.com.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-22:
@dgarske I have added `#define NO_FILESYSTEM`, but still same build error

dgarske, 2020-07-22:
Hi @geneva1 

Please send me your options.h and latest ./configure statement￼￼. If possible a link to your fork and branch with instructions for building. ￼

geneva1, 2020-07-23:
@dgarske Please find attached the [OP-TEE source code](https://drive.google.com/file/d/1fhoLF83PU-5fvq9wPnTo9Lfr__lsn6Ks/view?usp=sharing) with the [instructions](https://translate.google.com/translate?sl=auto&tl=en&u=https%3A%2F%2Fm.blog.naver.com%2FPostView.nhn%3FblogId%3Dhw5773%26logNo%3D221319208008%26proxyReferer%3Dhttps%3A%252F%252Fwww.google.com%252F)

Just run `make` inside `optee-qemu/build` to compile wolfssl for optee ARM architecture.

```
./configure --host=arm-linux CC=/home/rog/Downloads/optee-qemu/build/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc 
LD=/home/rog/Downloads/optee-qemu/build/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ld RANLIB=/home/rog/Downloads/optee-qemu/build/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ranlib 
AR=/home/rog/Downloads//optee-qemu/build/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf/bin/arm-linux-gnueabihf-ar
--enable-psk --enable-sni --enable-crl --enable-static --enable-savesession --enable-pkcallbacks --enable-memory --enable-dh 
--enable-aes --enable-poly1305 --enable-chacha --disable-examples --disable-shared --prefix=/tmp/wolfssl 
--enable-singlethreaded --enable-debug --enable-bigcache --enable-crl --enable-static --enable-savesession --enable-memory 
--enable-dh --enable-aes --enable-poly1305 --enable-chacha --enable-hugecache --enable-pkcallbacks --enable-aesccm 
--enable-hkdf --enable-ecc --enable-dsa --enable-ocsp --enable-tlsx --enable-sni --enable-single-threaded --disable-examples --disable-shared
```

dgarske, 2020-07-23:
Hi @geneva1 , 

Your tar/gz is incomplete (upload did not finish). I am however setting up OPTEE with QEMU to try and reproduce building and running wolfSSL. I am using Ubuntu 18.

Here are the OPTEE setup instructions so far:

```
mkdir optee-qemu
cd optee-qemu
repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml
repo sync -j4 --no-clone-bundle

cd build
make -j4 toolchains
make -j4
make run
```

Next I'll attempt to cross-compile wolfSSL for it.

Thanks,
David Garske, wolfSSL

geneva1, 2020-07-24:
@dgarske I have reuploaded the file again at https://github.com/wolfSSL/wolfssl/issues/3143#issuecomment-662846918 as well as sent the file to your email address : david@wolfssl.com

Note: I have confirmed that the tar.xz file is not corrupted before file upload/send.

geneva1, 2020-07-25:
@dgarske I think I know what is wrong. The tar.xz file is not compressed correctly even though it is not corrupted before file upload/send process. 

I will try to re-compress using zip compression format.

geneva1, 2020-07-27:
@dgarske okay, [this 7z file](https://drive.google.com/file/d/1HXWLYAw4cealfhkRjVKhxFD48xb0712C/view?usp=sharing) should download and extract just fine.

closed 2021-01-15 by haydenroche5