wolfSSL/wolfssl #3147: WolfSSL Does Not Work On Ubuntu 20.04 And Fails With Warning About Timing Attacks / Harden Optiosn (tested distro package and compile from source)

bonedaddy, 2020-07-17:
I'm attempting to build a project that's using WolfSSL but it is failing to compile with the following error:

```
In file included from /usr/local/include/wolfssl/wolfcrypt/types.h:29,
                 from /usr/local/include/wolfssl/wolfcrypt/sha.h:30,
                 from /media/solidity/DATA/Code/RTradeLtd/libcp2p/src/crypto/sha256.c:3:
/usr/local/include/wolfssl/wolfcrypt/settings.h:2124:14: error: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Werror=cpp]
 2124 |             #warning "For timing resistance / side-channel attack prevention consider using harden options"
      |              ^~~~~~~
```

I've gotten this error both from installing wolf via apt packages (`sudo apt install libwolfssl24 libwolfssl-dev`) as well as compiling WolfSSL from source. I [tried these instructions](https://www.wolfssl.com/forums/topic1171-building-library-on-ubuntu-1710.html) and still got the same error.

Here is how I tried to build from source:
```
./configure -enable-harden -enable-ecc -enable-keygen
make -j6
sudo make install -j6
```

# Version Information

* I've tried `WolfSSL` latest release, as well as master branch
* gcc version:  9.3.0
* kernel version: `Linux gray 5.4.0-7634-generic #38~1592497129~20.04~9a1ea2e-Ubuntu SMP Fri Jun 19 22:43:37 UTC  x86_64 x86_64 x86_64 GNU/Linux`

OS information
```
DISTRIB_ID=Pop
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Pop!_OS 20.04 LTS"
```

dgarske, 2020-07-17:
Hi @postables,

Make sure and include `#include <wolfssl/options.h>` in your application before any other wolf headers. That will pull in the ./configure settings and resolve this.

Thanks,
David Garske, wolfSSL

bonedaddy, 2020-07-17:
That makes sense, thanks.

closed 2020-07-17 by bonedaddy