wolfSSL/wolfssl #3213: Missing check or incorrect conversion of error codes at wolfSSL_write/read

ErrReporter, 2020-08-11:
https://github.com/wolfSSL/wolfssl/blob/4f30e370944498d024f1735f0ffd5c9cdfbaa858/src/ssl.c#L1895

In function `wolfSSL_read_internal`, the `ret` variable can be overwritten by return value of `NotifyWriteSide`. So this error is ignored.
```c
    ret = ReceiveData(ssl, (byte*)data, sz, peek);

#ifdef HAVE_WRITE_DUP
    if (ssl->dupWrite) {
        if (ssl->error != 0 && ssl->error != WANT_READ
        #ifdef WOLFSSL_ASYNC_CRYPT
            && ssl->error != WC_PENDING_E
        #endif
        ) {
            int notifyErr;

            WOLFSSL_MSG("Notifying write side of fatal read error");
            notifyErr  = NotifyWriteSide(ssl, ssl->error);
            if (notifyErr < 0) {
                ret = ssl->error = notifyErr;
            }
        }
    }
#endif

```

Even the error is correctly received, in the end, the original error code does not held at:
```c
if (ret < 0)
        return WOLFSSL_FATAL_ERROR;
```
It means all errors become a fatal error, which makes it hard to locate where is the problem. Shall we just keep error codes from `ReceiveData` or other APIs since they are defined errors from the [manual](https://www.wolfssl.com/docs/wolfssl-manual/appendix-c/)?

dgarske, 2020-08-11:
Thank you for this report. I've reviewed this code closely and this is by design. The `ReceiveData` function sets `ssl->error` in all cases. This section of code is used by the `HAVE_WRITE_DUP` feature, which only uses this code path to inform the write side of an error. If that code path is executed and the side notification fails then `ret` is overridden. Marking closed / won't fix.

ErrReporter, 2020-08-12:
> Thank you for this report. I've reviewed this code closely and this is by design. The `ReceiveData` function sets `ssl->error` in all cases. This section of code is used by the `HAVE_WRITE_DUP` [...]

@dgarske Thank you for your reply. I understand the overridden `ret`  is by design, but shall various errors be unified as `WOLFSSL_FATAL_ERROR`? From my manual review, the `ReceiveData`/`SendData` can return `WOLFSSL_FATAL_ERROR`, `WOLFSSL_CBIO_ERR_WANT_WRITE`, `BAD_FUNC_ARG`, and more. I really suggest keeping the original error than dropping them all by ```if (ret < 0) return WOLFSSL_FATAL_ERROR```, since you have already clearly differentiated the error types. Because, apparently, not all cases `ssl->error` are set consistently with return values. For example,

`src/internal.c` `SendData()`
```c
if (ssl->earlyData != no_early_data) {
        if (ssl->options.handShakeState == HANDSHAKE_DONE) {
            WOLFSSL_MSG("handshake complete, trying to send early data");
            return BUILD_MSG_ERROR;
        }
```

the return values `BUILD_MSG_ERROR` is different from `ssl->error`.

dgarske, 2020-08-12:
Hi @ErrReporter , 

Thanks again for the report. You are correct. I missed that place for early data and `BUILD_MSG_ERROR`. I will put up a PR for this and the other issue #3212.

Thanks, 
David Garske, wolfSSL