wolfSSL/wolfssl #3226: Port to vcpkg

wangwenx190, 2020-08-20:
wolfSSL uses CMake so I think it's not diffcult to port to vcpkg.

haydenroche5, 2021-01-14:
I have filed this request in our internal feature request list. If there's interest in having us port wolfSSL to vcpkg, please reach out to facts@wolfssl.com. Thanks!

closed 2021-01-14 by haydenroche5

Kiddinglife, 2021-08-21:
any comments on vcpkg ? now it is still not possible to install wolfssl via vcpkg.

elms, 2022-04-22:
@wangwenx190 I've update https://github.com/wolfSSL/wolfssl/issues/4323 to indicate that there is a [PR](https://github.com/microsoft/vcpkg/pull/24348) to add vcpkg support

wangwenx190, 2022-04-23:
> @wangwenx190 I've update #4323 to indicate that there is a [PR](https://github.com/microsoft/vcpkg/pull/24348) to add vcpkg support

Thanks! That's really good to hear!