wolfSSL/wolfssl #3240: Build failure on big endian platforms (e.g. IBM Z/s390x)

tobiasbrunner, 2020-08-24:
The [`#ifdef` around `ByteReverseWords`](https://github.com/wolfSSL/wolfssl/commit/cebb28382243707a435c12db370f1c1a1e4741e3#diff-4be40df89ec617072a3441685b231559R123) that was added with cebb28382243707a435c12db370f1c1a1e4741e3 and released with 4.5.0 breaks the build on big endian platforms (e.g. on s390x on Travis CI, see [here](https://travis-ci.org/github/strongswan/strongswan/jobs/719846820#L820)):
```
wolfcrypt/src/md5.c: In function ‘wc_Md5Update���:
wolfcrypt/src/md5.c:378:13: error: implicit declaration of function ‘ByteReverseWords’; did you mean ‘ByteReverseWords64’? [-Werror=implicit-function-declaration]
             ByteReverseWords(md5->buffer, md5->buffer, WC_MD5_BLOCK_SIZE);
             ^~~~~~~~~~~~~~~~
             ByteReverseWords64
wolfcrypt/src/md5.c:378:13: error: nested extern declaration of ‘ByteReverseWords’ [-Werror=nested-externs]
```

tmael, 2020-08-26:
Hi @tobiasbrunner and @kaleb-himes,

I've opened PR #3255 to fix the issue. 

Thanks

tobiasbrunner, 2020-08-31:
I successfully tested e9b1ceae7e1d. Thanks!

closed 2020-08-31 by tobiasbrunner