wolfSSL/wolfssl #3270: wolfssl renegotiation support

hihigupt, 2020-09-02:
Hi,

I am using curl with wolfssl to connect to https endpoint, but connection to the server is dropped via client. This is because server sends a renegotiation request and client does not know how to handle (as curl did not add support for renegotiation, please look at the issue I filed on curl https://github.com/curl/curl/issues/5839). 

To address runtime support, I added code to enable renegotiation https://github.com/hihigupt/curl/blob/52e7dcf7ac8fb88635cb9a2b7d6d60db0bb64c06/lib/vtls/wolfssl.c#L502.

But I am still seeing the connection is closed via client, and if I enable logs of wolf ssl I am seeing following log before alert sent to server:

* STATE: PROTOCONNECT => DO handle 0x55c92aa0ae38; line 2028 (connection #1)
wolfSSL Entering SSL_write()
growing output buffer

wolfSSL Entering BuildMessage
wolfSSL Leaving BuildMessage, return 0
Shrinking output buffer

wolfSSL Leaving SSL_write(), return 221
> POST /devices/test88/messages/events?api-version=2016-11-14 HTTP/1.1
> Host: kartos.azure-devices.net
> User-Agent: curl/7.73.0-DEV
> Accept: */*
> Content-Type:application/json
> Content-Length: 24
>
* upload completely sent off: 24 out of 24 bytes
* STATE: DO => DO_DONE handle 0x55c92aa0ae38; line 2083 (connection #1)
* STATE: DO_DONE => PERFORM handle 0x55c92aa0ae38; line 2204 (connection #1)
wolfSSL Entering wolfSSL_read()
wolfSSL Entering wolfSSL_read_internal()
wolfSSL Entering ReceiveData()
growing input buffer

received record layer msg
wolfSSL Entering DoHandShakeMsg()
wolfSSL Entering DoHandShakeMsgType
processing hello request
wolfSSL Entering DoHelloRequest
wolfSSL Leaving DoHelloRequest, return 0
Shrinking input buffer

wolfSSL Leaving DoHandShakeMsgType(), return 0
Need to start scr, server requested
**Can't renegotiate until previous handshake complete**
wolfSSL Leaving wolfSSL_read_internal(), return -388
wolfSSL Entering SSL_get_error
wolfSSL Leaving SSL_get_error, return 0
wolfSSL Entering ERR_error_string

Is there anything else missing from my side or renegotiation feature is broken? I will attaching wireshark trace and key to this issue.

[github_wireshark.zip](https://github.com/wolfSSL/wolfssl/files/5164417/github_wireshark.zip)

dgarske, 2020-09-08:
Hi @hihigupt ,

Typically "wolfSSL_UseSecureRenegotiation" is disabled because it is not "secure" as implied and is typically only used with IIS / Windows Servers. Are you positive this is the feature you are trying to use? Perhaps you mean "resumption" and reusing a previous connection?

To use the Secure Renegotiation you need to build wolfSSL with `HAVE_SECURE_RENEGOTIATION` or "--enable-secure-renegotiation" and call `wolfSSL_UseSecureRenegotiation` or `wolfSSL_CTX_UseSecureRenegotiation`.

Example:
https://github.com/wolfSSL/wolfssl/blob/master/examples/client/client.c#L3309

To use session resumption:

```
session = wolfSSL_get_session(ssl);
/* close TLS / socket */

wolfSSL_set_session(sslResume, session);
```

To use Session Tickets:

```
#ifdef HAVE_SESSION_TICKET
        wolfSSL_set_SessionTicket_cb(sslResume, sessionTicketCB,
                                    (void*)"resumed session");
#endif
```

Thanks,
David Garske, wolfSSL

hihigupt, 2020-09-08:
Hi @dgarske 

I am using curl to connect to https endpoint in Azure  for Azure IotHub service. Typical flow for client to connect to server on https protocol is: client connects to server using anonymous tls session (by not asking any client certificate). Once, tls session is established, client sends a request to target URL as a HTTP request in the SSL tunnel. Now if target URL requires or at least demands certificate-based client authentication, the server triggers a new handshake (renegotiation). Current implementation of curl with wolf ssl does not have any support (both compile and runtime) for renegotiation. I tried adding this support in curl (branch https://github.com/hihigupt/curl/blob/52e7dcf7ac8fb88635cb9a2b7d6d60db0bb64c06/lib/vtls/wolfssl.c#L502, both compile and runtime), but it still fails to create tls session after renegotiation triggered by server. Looking further into wolf ssl code, it looks like some issue with wolf ssl state not being correct (at-least base on the logs, when enable for wolf ssl). Please let me know if the API that are good enough to enable the renegotiation.

Note: Steps I used for building curl with wolfssl

cd /home/hihigupt/gitroot/wolfssl/
./configure --prefix=/home/hihigupt/wolfsslinstall --enable-postauth --enable-tls13 --enable-libwebsockets --enable-harden --enable-aesni --enable-psk --enable-stacklog --enable-sni --enable-alpn --enable-trustedca --enable-truncatedhmac --enable-stunnel --enable-certservice --enable-enckeys --enable-secure-renegotiation --enable-debug
make
make install

cd /home/hihigupt/gitroot/curl
./configure --with-wolfssl=/home/hihigupt/wolfsslinstall --without-ssl --without-zlib --enable-debug
make
make install

Wolf ssl Logs:

wolfSSL Leaving DoHandShakeMsgType(), return 0
Need to start scr, server requested
**Can't renegotiate until previous handshake complete**
wolfSSL Leaving wolfSSL_read_internal(), return -388
wolfSSL Entering SSL_get_error
wolfSSL Leaving SSL_get_error, return 0
wolfSSL Entering ERR_error_string

dgarske, 2020-09-08:
Hi @hihigupt ,

I'd like to escalate this internally. Would you mind sending a direct email to support@wolfssl.com?

Thanks,
David Garske, wolfSSL

hihigupt, 2020-09-08:
Hi @dgarske 

Thanks !

I have sent an email to wolfssl support.

dgarske, 2020-09-21:
Hi @hihigupt , we are tacking this feature request via a support ticket, so marking this closed. Thanks

closed 2020-09-21 by dgarske