wolfSSL/wolfssl #3284: aes in ctr mode is not using intel aes-ni (other modes do)

cryptgithub, 2020-09-09:
without aes-ni
in wolfssl's own benchmark tool
cbc and ctr mode were getting about 150-200mb/s, and gcm about 15-20mb/s   

after enabling aes-ni
i got about 1000-9000mb/s in cbc and gcm mode but still the same 150-200mb in ctr 

question 1:
i wonder why ctr is much faster than gcm in software considering that gcm uses ctr 
bug or misconfiguration :
 why enabling aes-ni doesn't effect ctr mode ,when it does GCM that uses CTR internally 

my setup :
os :windows server 2019 inside vmware (host ubuntu )
cpu : amd ryzen 9 3900x
compiler : visual stdio 2019

here's output example 
wolfCrypt Benchmark (block bytes 1048576, min 1.0 sec each)
RNG                110 MB took 1.037 seconds,  106.053 MB/s
AES-128-CBC-enc   1300 MB took 1.001 seconds, 1298.371 MB/s
AES-128-CBC-dec   9160 MB took 1.001 seconds, 9154.723 MB/s
AES-192-CBC-enc   1175 MB took 1.001 seconds, 1174.393 MB/s
AES-192-CBC-dec   7840 MB took 1.001 seconds, 7833.868 MB/s
AES-256-CBC-enc   1080 MB took 1.004 seconds, 1075.887 MB/s
AES-256-CBC-dec   6875 MB took 1.001 seconds, 6870.924 MB/s
AES-128-GCM-enc   3340 MB took 1.001 seconds, 3336.536 MB/s
AES-128-GCM-dec   3130 MB took 1.001 seconds, 3125.852 MB/s
AES-192-GCM-enc   3085 MB took 1.000 seconds, 3084.334 MB/s
AES-192-GCM-dec   3065 MB took 1.002 seconds, 3060.332 MB/s
AES-256-GCM-enc   2965 MB took 1.001 seconds, 2962.409 MB/s
AES-256-GCM-dec   3075 MB took 1.000 seconds, 3074.216 MB/s
AES-128-CTR        215 MB took 1.014 seconds,  212.059 MB/s
AES-192-CTR        185 MB took 1.007 seconds,  183.661 MB/s
AES-256-CTR        165 MB took 1.008 seconds,  163.617 MB/s
ARC4               350 MB took 1.003 seconds,  348.784 MB/s
RABBIT             725 MB took 1.002 seconds,  723.721 MB/s
3DES                35 MB took 1.023 seconds,   34.218 MB/s
MD5                600 MB took 1.002 seconds,  598.794 MB/s
SHA                645 MB took 1.006 seconds,  641.275 MB/s
SHA-256            275 MB took 1.002 seconds,  274.491 MB/s
SHA-384            355 MB took 1.009 seconds,  351.761 MB/s
SHA-512            345 MB took 1.012 seconds,  341.075 MB/s
RIPEMD             265 MB took 1.001 seconds,  264.729 MB/s
HMAC-MD5           620 MB took 1.001 seconds,  619.448 MB/s
HMAC-SHA           620 MB took 1.000 seconds,  619.954 MB/s
HMAC-SHA256        285 MB took 1.016 seconds,  280.515 MB/s
HMAC-SHA384        345 MB took 1.005 seconds,  343.125 MB/s
HMAC-SHA512        330 MB took 1.002 seconds,  329.253 MB/s
PBKDF2              34 KB took 1.001 seconds,   34.252 KB/s
RSA     2048 public       7900 ops took 1.002 sec, avg 0.127 ms, 7886.350 ops/sec
RSA     2048 private       200 ops took 1.426 sec, avg 7.132 ms, 140.213 ops/sec
DH      2048 key gen       348 ops took 1.000 sec, avg 2.875 ms, 347.865 ops/sec
DH      2048 agree         400 ops took 1.167 sec, avg 2.918 ms, 342.750 ops/sec
ECC      256 key gen      6110 ops took 1.000 sec, avg 0.164 ms, 6109.213 ops/sec
ECDHE    256 agree        2400 ops took 1.010 sec, avg 0.421 ms, 2377.289 ops/sec
ECDSA    256 sign         4900 ops took 1.015 sec, avg 0.207 ms, 4825.926 ops/sec
ECDSA    256 verify       2100 ops took 1.026 sec, avg 0.489 ms, 2046.852 ops/sec
ED     25519 key gen     27270 ops took 1.000 sec, avg 0.037 ms, 27269.414 ops/sec
ED     25519 sign        25500 ops took 1.003 sec, avg 0.039 ms, 25432.865 ops/sec
ED     25519 verify       9700 ops took 1.007 sec, avg 0.104 ms, 9628.677 ops/sec
Benchmark complete 

my user_setting.h in
[user_setting.h.txt](https://github.com/wolfSSL/wolfssl/files/5195312/user_setting.h.txt)
 attachment 

thanks in advance

dgarske, 2020-09-09:
Hi @cryptgithub ,

Thanks for this report and feature request. I was able to reproduce these results on my Mac. AES CBC, GCM, ECB and CCM are AESNI accelerated, but not AES CTR. Looks like a simple issue with `aes->use_aesni` not getting set... I will put up a fix and link it here shortly.

Thanks,
David Garske, wolfSSL

cryptgithub, 2020-09-10:
thanks for the fix 
but still very slow compared to both gcm (witch is ctr+gmac) and other libraries like crypto++

in my machine here's resualt after your fix 
AES-128-GCM-enc   3195 MB took 1.001 seconds, 3190.752 MB/s
AES-128-GCM-dec   3080 MB took 1.002 seconds, 3075.338 MB/s
AES-192-GCM-enc   2615 MB took 1.001 seconds, 2613.301 MB/s
AES-192-GCM-dec   2855 MB took 1.001 seconds, 2853.432 MB/s
AES-256-GCM-enc   2910 MB took 1.001 seconds, 2908.025 MB/s
AES-256-GCM-dec   2975 MB took 1.002 seconds, 2970.367 MB/s
AES-128-CTR        690 MB took 1.002 seconds,  688.358 MB/s
AES-192-CTR        590 MB took 1.004 seconds,  587.743 MB/s
AES-256-CTR        570 MB took 1.007 seconds,  565.971 MB/s

you see gcm that should be slower (ctr+gmac) is 5x faster still, that's huge gap

also the same machine using crypto++:
AES/CTR (128-bit key)	AESNI	2111	0.156
AES/CTR (192-bit key)	AESNI	1739	0.157
AES/CTR (256-bit key)	AESNI	1480	0.165 

still 3x slower

i'm not cryptographer my understanding gcm is ctr + gmac with a small difference in  initializing iv (witch should not slow down )i ,so it should be slower than ctr 
is there some thing wrong in the code or i'm wrong ?

dgarske, 2020-09-10:
Hi @cryptgithub ,

Thanks for those comparison numbers. I agree AES-CTR with AESNI could be improved. I suspect we've focused on speeds for AES CBC and GCM because they are used for TLS.

To help track this feature request perhaps you can email support@wolfssl.com with and tell us a bit more about your project?

Thanks,
David Garske, wolfSSL

cryptgithub, 2020-09-11:
ok than i think i will use GCM and throw away the mac 
additional info unrelated to this issue:
my project is simple free windows data backup software, it encrypts when backing up data, so didn't want to slow down for encryption
choosed your library because i guess you should be more focused on secure implementation as you have hash based RNG  , now i see most of the code is not event checked once

thanks for your time and fix

dgarske, 2020-09-21:
Thanks @cryptgithub for those details on your project and the positive feedback regarding our library. The fix was merged in PR #3286, so marking this ticket as solved.

closed 2020-09-21 by dgarske