wolfSSL/wolfssl #3292: "early_data" length is 0

xielizyh, 2020-09-12:
Hi,
   When sending early data with using PSK, the early_data extension can be found, but its length is 0. So the peer(server) can not parse and ignore the early data.  The capture with Wireshark is as follows:
[image]

SparkiDev, 2020-09-13:
I'm not sure what the issue you are raising is.

RFC 8446, Section 4.2.10:

      struct {} Empty;

      struct {
          select (Handshake.msg_type) {
              case new_session_ticket:   uint32 max_early_data_size;
              case client_hello:         Empty;
              case encrypted_extensions: Empty;
          };
      } EarlyDataIndication;

The early data extension is 0 length in the ClientHello.
Early data is sent as application data.

Looking at the TLS 1.3 code, the early data extension should only be send if wolfSSL_write_early_data() is called. That is early data is being explicitly sent by the client application.

Could you please provide more information?

Thanks,
Sean

xielizyh, 2020-09-14:
Thanks for your prompt response. 

I'm sorry I had a wrong understanding of the early data mechanism.  I mistakenly thought the early data should be inside in the ClientHello. In fact, the client can send early data as Application Data in its first flight of messages.
[image]

Thanks again for your patience.

SparkiDev, 2020-09-14:
Glad we could help you understand.

Sean :-)

closed 2020-09-14 by SparkiDev