wolfSSL/wolfssl #3293: "WOLFSSL_FFDHE_3072~WOLFSSL_FFDHE_8192" groups report an error

xielizyh, 2020-09-14:
Hi,
   when using wolfSSL_UseKeyShare(ssl, WOLFSSL_FFDHE_3072), it return a error code, that is "-112". Is it meanning that it is mow unable to use DH 3072-bit parameters?

Thanks in advance.

dgarske, 2020-09-14:
Hi @xielizyh ,

Make sure you have FP_MAX_BITS defined to be the max RSA/DH key size times 2. So for 3072 bit DH you should add `CFLAGS="-DFP_MAX_BITS=6144` or `#define FP_MAX_BITS 6144`. 

Also with TLS 1.3 you need to define the DH key parameters you'd like to include using `-DHAVE_FFDHE_3072` or `#define `HAVE_FFDHE_3072`.

Thanks,
David Garske, wolfSSL

xielizyh, 2020-09-15:
Hi @garske,

Follow your settings, the FFDHE-3072~FFDHE-8192 groups were all successfully implemented. 

[image]

Thanks.

dgarske, 2020-09-15:
You are welcome. Marking issue closed.

closed 2020-09-15 by dgarske