wolfSSL/wolfssl #3301: DH6144 or DH8192 key is all '0'

xielizyh, 2020-09-17:
Hi,
   It's me again. When using ```HAVE_FFDHE_6144``` or ```HAVE_FFDHE_8192```, DH key value is generated all '0', as follows:

[image]

Thanks in advance.

SparkiDev, 2020-09-17:
Hi @xielizyh 

I've not been able to reproduce this.

I used the configuration line:
  ./configure C_EXTRA_FLAGS="-DWOLFSSL_FFDHE_6144 -DWOLFSSL_FFHDE_8192 -DFP_MAX_BITS=16384 -DWOLFSSL_DEBUG_TLS" --enable-debug

This allows you to see the TLS 1.3 handshake data.
When I ran the wolfSSL client and server, the generated public key was not all zeros and the data sent in the TLS packets was not all zeros.

Please let me know what difference in configuration you used that results in the above.

Sean

xielizyh, 2020-09-18:
Hi @SparkiDev 

I ran the TLS1.3 client in the esp32(espressif) with using config as follows:
```
#define FP_MAX_BITS 16384
#define HAVE_DH
#define HAVE_FFDHE
#define HAVE_FFDHE_2048
#define HAVE_FFDHE_3072
#define HAVE_FFDHE_4096
#define HAVE_FFDHE_6144
#define HAVE_FFDHE_8192
```

I found that the all zeros public key was caused by the following code in `tls.c`: 
```
    if (params->p_len != dataSz) {
        /* Zero pad the front of the public key to match prime "p" size */
        XMEMMOVE(keyData + params->p_len - dataSz, keyData, dataSz);
        XMEMSET(keyData, 0, params->p_len - dataSz);
    }
```

The debug log is that:
```
[11:47:03.310]I (11938) tls: Start to run tls client
[11:47:03.324]I (11938) tls: The current date/time(dummy) is: Sat Aug 22 02:44:54 2020
[11:47:03.325]I (11938) tls: Debug on
[11:47:03.329]I (11948) tls: Create a socket...
[11:47:03.337]I (11948) tls: Socket created, connecting to 114.55.142.4:10000...
[11:47:03.687]esp32> I (12298) tls: Connected to server successfully
[11:47:03.689]I (12298) tls: Initialize wolfSSL
[11:47:03.696]I (12298) wolfssl: wolfSSL Entering wolfSSL_Init
[11:47:03.697]I (12318) wolfssl: wolfSSL Entering wolfCrypt_Init
[11:47:03.704]I (12318) tls: Create and initialize wolfSSL_CTX
[11:47:03.706]I (12318) wolfssl: wolfSSL Entering TLSv1_3_client_method_ex
[11:47:03.712]I (12328) wolfssl: wolfSSL Entering wolfSSL_CTX_new_ex
[11:47:03.716]I (12338) wolfssl: wolfSSL Entering wolfSSL_CertManagerNew
[11:47:03.721]I (12338) wolfssl: wolfSSL Leaving WOLFSSL_CTX_new, return 0
[11:47:03.733]I (12348) wolfssl: wolfSSL Entering wolfSSL_CTX_load_verify_buffer_ex
[11:47:03.734]I (12348) wolfssl: Adding a CA
[11:47:03.741]I (12348) wolfssl: wolfSSL Entering GetExplicitVersion
[11:47:03.745]I (12358) wolfssl: wolfSSL Entering GetSerialNumber
[11:47:03.751]I (12358) wolfssl: Got Cert Header
[11:47:03.752]I (12378) wolfssl: wolfSSL Entering GetAlgoId
[11:47:03.760]I (12378) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.761]I (12388) wolfssl: Got Algo ID
[11:47:03.780]I (12388) wolfssl: Getting Cert Name
[11:47:03.780]I (12388) wolfssl: Getting Cert Name
[11:47:03.781]I (12398) wolfssl: Got Subject Name
[11:47:03.789]I (12398) wolfssl: wolfSSL Entering GetAlgoId
[11:47:03.791]I (12398) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.797]I (12408) wolfssl: Got Key
[11:47:03.798]I (12408) wolfssl: Parsed Past Key
[11:47:03.805]I (12408) wolfssl: wolfSSL Entering DecodeCertExtensions
[11:47:03.807]I (12418) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.814]I (12428) wolfssl: wolfSSL Entering DecodeSubjKeyId
[11:47:03.815]I (12438) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.824]I (12438) wolfssl: wolfSSL Entering DecodeAuthKeyId
[11:47:03.825]I (12448) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.832]I (12448) wolfssl: wolfSSL Entering DecodeBasicCaConstraint
[11:47:03.833]I (12458) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.840]I (12458) wolfssl: wolfSSL Entering DecodeAltNames
[11:47:03.845]I (12468) wolfssl:      Unsupported name type, skipping
[11:47:03.861]I (12468) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.862]I (12478) wolfssl: DecodeExtKeyUsage
[11:47:03.870]I (12478) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.872]I (12478) wolfssl: wolfSSL Entering GetObjectId()
[11:47:03.877]I (12488) wolfssl: wolfSSL Entering GetAlgoId
[11:47:03.884]I (12488) wolfssl: wolfSSL Entering GetObjectId()
[... 148 lines cut ...]
[11:47:05.107]I (13728) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.119]I (13728) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.125]I (13738) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.137]I (13748) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.138]I (13758) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.150]I (13758) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.156]I (13768) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.170]I (13778) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.172]I (13788) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.179]I (13798) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.190]I (13808) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.200]I (13818) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.208]I (13818) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.218]I (13828) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.223]I (13838) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.228]I (13848) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.240]I (13858) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.250]I (13868) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.263]I (13878) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.264]I (13878) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.276]I (13888) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.283]I (13898) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[11:47:05.289]I (13908) tls: Set ffdhe-8192 successfully
[11:47:05.298]I (13908) tls: Connecting to tls server... 
```

SparkiDev, 2020-09-20:
Hi @xielizyh
Thanks for digging into this.

I can't see the issue with the code.

When using FFDHE8192:
params->p_len = 1024
dataSz will be less than 1024 to enter this:

    if (params->p_len != dataSz) {
        /* Zero pad the front of the public key to match prime "p" size */
        XMEMMOVE(keyData + params->p_len - dataSz, keyData, dataSz);
        XMEMSET(keyData, 0, params->p_len - dataSz);
    }

The code moves all the data to the end of the 1024 byte data (assuming XMEMMOVE is working), then sets the front unused bytes to zero.

What is dataSz?
Is the XMEMMOVE working for you or does it make the bytes all zeros?

Sean

xielizyh, 2020-09-21:
Hi @SparkiDev 

I found that the XMEMMOVE worked normaly. Furthermore，it may be caused by `wc_DhGenerateKeyPair`. The function could generate private key, but not public key with using FFDHE_6144 or FFDHE_8192. I print the priv_key and public_key as follows: 

``[10:27:25.767]I (72536) tls: Set key share...
[10:27:25.837]I (72626) wc_DhGenerateKeyPair: priv[29]:
[10:27:25.837]7c 7e 3c 4b 23 6c 71 8c 15 7d b3 4f ca 75 74 86 
[10:27:25.847]be 57 b3 d8 9a 30 dc b8 66 2b e4 2f da 
[10:27:25.847]I (72626) wc_DhGenerateKeyPair: 
[10:27:25.847]pub[256]:
[10:27:25.857]3f be e5 87 d5 77 24 40 e4 d2 d8 69 bc 34 fd e4 
[10:27:25.867]a4 cd 85 9e 3a bd 9f cc 68 4f 1a 8b 72 e7 2a 11 
[10:27:25.867]46 7e 2c 90 bc 3c e7 70 98 99 fa fc 02 2b 01 49 
[10:27:25.877]b6 f4 60 d3 6c 3e f8 e4 c1 2f 9c bd 9d 9e c9 26 
[10:27:25.877]e4 8d c3 b8 73 b6 9f b9 94 28 2c 48 2c f8 0c a3 
[10:27:25.877]d0 25 fb 10 bf 1d 1c 03 37 c4 49 88 5b 7c f8 88 
[10:27:25.887]62 c5 00 3b ca d9 06 f4 dd 49 47 9c e7 c3 b2 93 
[10:27:25.887]4c 76 32 e8 a9 96 c4 3b 0f 6b 60 02 ca bc 81 05 
[10:27:25.897]9e 55 4c fe 52 e5 2b 0c 0d 41 7e 43 f5 75 69 39 
[10:27:25.897]3e eb e5 e5 9c fb 12 eb b3 6f b5 df 62 57 a7 ba 
[10:27:25.897]c5 7e 0a 4b 0b e3 d3 49 d6 65 54 ff 56 69 68 79 
[10:27:25.907]f4 85 97 94 3f c6 59 95 83 33 84 76 c9 c4 a1 6d 
[10:27:25.907]9c 77 06 a9 0f 7c d4 50 a7 0a c7 b8 2f 38 8b 7f 
[10:27:25.917]c1 be 66 ee 41 af 94 fb a0 58 86 4a 8a 58 0e a1 
[10:27:25.917]20 56 dd 99 ff ad 05 7a fb 75 b0 88 fa e9 22 50 
[10:27:25.917]01 70 e0 52 bb 38 71 fc 2f e8 61 66 a1 65 12 7f 
[10:27:25.917]
[10:27:25.927]I (72716) wolfssl: Public DH Key
[10:27:25.927]I (72716) wolfssl:      3f be e5 87 d5 77 24 40 e4 d2 d8 69 bc 34 fd e4 |?....w$@...i.4..
[10:27:25.937]I (72726) wolfssl:      a4 cd 85 9e 3a bd 9f cc 68 4f 1a 8b 72 e7 2a 11 |....:...hO..r.*.
[10:27:25.947]I (72726) wolfssl:      46 7e 2c 90 bc 3c e7 70 98 99 fa fc 02 2b 01 49 |F~,..<.p.....+.I
[10:27:25.957]I (72736) wolfssl:      b6 f4 60 d3 6c 3e f8 e4 c1 2f 9c bd 9d 9e c9 26 |..`.l>.../.....&
[10:27:25.977]I (72746) wolfssl:      e4 8d c3 b8 73 b6 9f b9 94 28 2c 48 2c f8 0c a3 |....s....(,H,...
[10:27:25.977]I (72756) wolfssl:      d0 25 fb 10 bf 1d 1c 03 37 c4 49 88 5b 7c f8 88 |.%......7.I.[|..
[10:27:25.987]I (72756) wolfssl:      62 c5 00 3b ca d9 06 f4 dd 49 47 9c e7 c3 b2 93 |b..;.....IG.....
[10:27:25.997]I (72776) wolfssl:      4c 76 32 e8 a9 96 c4 3b 0f 6b 60 02 ca bc 81 05 |Lv2....;.k`.....
[10:27:26.007]I (72786) wolfssl:      9e 55 4c fe 52 e5 2b 0c 0d 41 7e 43 f5 75 69 39 |.UL.R.+..A~C.ui9
[10:27:26.007]I (72786) wolfssl:      3e eb e5 e5 9c fb 12 eb b3 6f b5 df 62 57 a7 ba |>........o..bW..
[10:27:26.017]I (72796) wolfssl:      c5 7e 0a 4b 0b e3 d3 49 d6 65 54 ff 56 69 68 79 |.~.K...I.eT.Vihy
[10:27:26.027]I (72806) wolfssl:      f4 85 97 94 3f c6 59 95 83 33 84 76 c9 c4 a1 6d |....?.Y..3.v...m
[10:27:26.037]I (72816) wolfssl:      9c 77 06 a9 0f 7c d4 50 a7 0a c7 b8 2f 38 8b 7f |.w...|.P..../8..
[10:27:26.037]I (72816) wolfssl:      c1 be 66 ee 41 af 94 fb a0 58 86 4a 8a 58 0e a1 |..f.A....X.J.X..
[10:27:26.047]I (72826) wolfssl:      20 56 dd 99 ff ad 05 7a fb 75 b0 88 fa e9 22 50 | V.....z.u...."P
[10:27:26.077]I (72846) wolfssl:      01 70 e0 52 bb 38 71 fc 2f e8 61 66 a1 65 12 7f |.p.R.8q./.af.e..
[10:27:26.077]I (72856) tls: Set ffdhe-2048 successfully
[10:27:26.087]I (72876) wc_DhGenerateKeyPair: priv[46]:
[10:27:26.087]4c 44 96 22 3c 00 5b 94 04 4b 51 af 06 77 2a 63 
[10:27:26.097]d0 ea ea 96 6b 4b fa 20 6c a5 aa 08 dd c5 a6 04 
[10:27:26.097]02 0e 18 6c 1b 23 18 e1 be 16 00 ea e5 98 
[10:27:26.097]I (72886) wc_DhGenerateKeyPair: 
[10:27:26.097]pub[764]:
[10:27:26.107]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.107]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.117]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.117]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.117]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.127]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.127]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.137]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.137]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.137]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.147]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.157]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.157]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.157]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.167]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.177]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.177]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.187]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10:27:26.187]00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... 192 lines cut ...]
[10:27:27.406]I (74186) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:27.417]I (74196) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.677]I (74206) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74206) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74216) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74226) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[... cut ...]
[10:27:30.687]I (74336) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74346) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74356) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74356) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74366) wolfssl:      00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 |................
[10:27:30.687]I (74386) tls: Set ffdhe-8192 successfully
```

SparkiDev, 2020-09-21:
If you are using wolfSSL on Espressif, are you using the files in: wolfssl/wolfcrypt/src/port/Espressif
If so, the esp32_mp.c:46-47 suggests the highest bit length for public operations is 2048.
That is only RSA and DH 2048 will work.

Let me know if this helps.

Sean

xielizyh, 2020-09-22:
Hi @SparkiDev

I have solved the problem. The reason  is that ESP32 RSA Accelerator supports operand lengths of N max to 4096 bits. So I did not use hw rsa primitive, ie:
'''#define NO_WOLFSSL_ESP32WROOM32_CRYPT_RSA_PRI"""

The capture of FFDHE_8192 with Wireshark is as follows:
[image]

So you can close this issue and thanks again for your patience.

SparkiDev, 2020-09-22:
Great news!
Glad I could help you through this!

Sean :-)

closed 2020-09-22 by SparkiDev