wolfSSL/wolfssl #3302: Build with picotcp failed

hckuo, 2020-09-17:
wolfssl version: 4.5.0
pico version (latest): https://github.com/tass-belgium/picotcp/commit/46120abecc9fb79f7dfb2cc8192341a42e40fc8b
How I configure wolfssl:
```
./configure \
    CFLAGS="-I$HOME/Developer/picotcp/build/include -Wno-error" \
    LDFLAGS="-L$HOME/Developer/picotcp/build/lib" \
```
wolfssl/wolfcrypt/settings.h
```
/* settings.h
 *
 * Copyright (C) 2006-2020 wolfSSL Inc.
 *
 * This file is part of wolfSSL.
 *
 * wolfSSL is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * wolfSSL is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1335, USA
 */

/* Place OS specific preprocessor flags, defines, includes here, will be
   included into every file because types.h includes it */

#ifndef WOLF_CRYPT_SETTINGS_H
#define WOLF_CRYPT_SETTINGS_H

#ifdef __cplusplus
    extern "C" {
#endif

/* This flag allows wolfSSL to include options.h instead of having client
 * projects do it themselves. This should *NEVER* be defined when building
 * wolfSSL as it can cause hard to debug problems. */
#ifdef EXTERNAL_OPTS_OPENVPN
#include <wolfssl/options.h>
#endif

/* Uncomment next line if using IPHONE */
/* #define IPHONE */

/* Uncomment next line if using ThreadX */
/* #define THREADX */

/* Uncomment next line if using Micrium uC/OS-III */
/* #define MICRIUM */

/* Uncomment next line if using Deos RTOS*/
/* #define WOLFSSL_DEOS*/

/* Uncomment next line if using Mbed */
/* #define MBED */

/* Uncomment next line if using Microchip PIC32 ethernet starter kit */
/* #define MICROCHIP_PIC32 */

/* Uncomment next line if using Microchip TCP/IP stack, version 5 */
/* #define MICROCHIP_TCPIP_V5 */

/* Uncomment next line if using Microchip TCP/IP stack, version 6 or later */
[... 2506 lines cut ...]
  377 |                 #define XFREE(p, h, t)       {void* xp = (p); if((xp)) wolfSSL_Free((xp));}
      | 
In file included from wolfcrypt/src/hash.c:27:
./wolfssl/wolfcrypt/settings.h:461: note: this is the location of the previous definition
  461 |     #define XFREE(p, h, type)    PICO_FREE((p))
      | 
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4515: wolfcrypt/src/src_libwolfssl_la-sha256.lo] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:5278: examples/server/tests_unit_test-server.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4501: wolfcrypt/src/src_libwolfssl_la-hmac.lo] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4508: wolfcrypt/src/src_libwolfssl_la-random.lo] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4613: wolfcrypt/src/src_libwolfssl_la-hash.lo] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:5264: examples/client/tests_unit_test-client.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4386: wolfcrypt/benchmark/benchmark.o] Error 1
cc1: all warnings being treated as errors
make[1]: *** [Makefile:5292: wolfcrypt/test/testsuite_testsuite_test-test.o] Error 1
make[1]: Leaving directory '/home/hckuo/Developer/wolfssl-4.5.0'
make: *** [Makefile:3291: all] Error 2
```

elms, 2020-09-18:
@hckuo I was able to compile with a change to settings and disabling examples. But running the tests fail on random.

Building examples fail, but I found some configure parameters to build some of them.

I built picotcp `make ARCH=shared`

```diff --git a/wolfssl/wolfcrypt/settings.h b/wolfssl/wolfcrypt/settings.h
index b683e73b0..2e57510b3 100644
--- a/wolfssl/wolfcrypt/settings.h
+++ b/wolfssl/wolfcrypt/settings.h
@@ -136,10 +136,10 @@
 /* #define WOLFSSL_TIRTOS */
 
 /* Uncomment next line if building with PicoTCP */
-/* #define WOLFSSL_PICOTCP */
+#define WOLFSSL_PICOTCP
 
 /* Uncomment next line if building for PicoTCP demo bundle */
-/* #define WOLFSSL_PICOTCP_DEMO */
+#define WOLFSSL_PICOTCP_DEMO
 
 /* Uncomment next line if building for uITRON4  */
 /* #define WOLFSSL_uITRON4 */
@@ -455,11 +455,18 @@
 
 #ifdef WOLFSSL_PICOTCP_DEMO
     #define WOLFSSL_STM32
+
+    #undef USE_FAST_MATH
     #define USE_FAST_MATH
+
+    #undef TFM_TIMING_RESISTANT
     #define TFM_TIMING_RESISTANT
+    #undef SINGLE_THREADED
+    #define SINGLE_THREADED
+
+    #define XMALLOC_OVERRIDE
     #define XMALLOC(s, h, type)  PICO_ZALLOC((s))
     #define XFREE(p, h, type)    PICO_FREE((p))
-    #define SINGLE_THREADED
     #define NO_WRITEV
     #define WOLFSSL_USER_IO
     #define NO_DEV_RANDOM
```

Building without examples
-------------------------

`./configure CFLAGS="-I$PWD/../picotcp/build/include -Wno-error" LDFLAGS="-L$PWD/../picotcp/build/lib -lpicotcp" --disable-examples`
:heavy_check_mark:

```
wolfcrypt/test/testwolfcrypt 
------------------------------------------------------------------------------
 wolfSSL version 4.5.0
------------------------------------------------------------------------------
error    test passed!
MEMORY   test passed!
base64   test passed!
RANDOM   test failed!
 error = -6700
```
:x: 

Building examples
-----------------

`./configure CFLAGS="-I$PWD/../picotcp/build/include -Wno-error" LDFLAGS="-L$PWD/../picotcp/build/lib -lpicotcp" --disable-asn --disable-rsa --enable-psk --disable-ecc`
:heavy_check_mark:

kaleb-himes, 2020-09-18:
@danielinux,

Any suggestions we might have for building with picoTCP that could help out @hckuo on this issue?

- KH

danielinux, 2020-09-19:
@hckuo what is the target for this build? A little more information on the application, operating system (if any), and hardware might help narrowing the scope. Older examples with wolfSSL+picoTCP exists on STM32/baremetal. In some cases it might help using a custom "user_settings.h".

The most recent embedded example is perhaps a TLS1.3 webserver on FreeRTOS/K64 to run remote firmware updates, available here:
https://github.com/wolfSSL/wolfBoot-examples/tree/master/freeRTOS-Freescale-K64F-https-TLS1.3

hckuo, 2020-09-19:
The target is x86_64 on a normal Linux box. The picoTCP uses a tap to send raw Ethernet frames.

danielinux, 2020-09-21:
@hckuo - [here is a pthread-based example for this scenario](https://github.com/danielinux/wolfssl-examples/tree/picotcp/picotcp)

It does not require any special wolfSSL compile-time configuration, since it implements custom I/O via `wolfSSL_CTX_SetIORecv`/`wolfSSL_CTX_SetIOSend` at runtime.

With this design, picoTCP "ticks" in a separate pthread, and all the socket calls are event-based, so a semaphore is used to synchronize transport rx events. Different (blocking) approaches are possible by attaching picoTCP to a BSD-socket layer.

hckuo, 2020-09-22:
@danielinux  Thank you for making this super helpful example! I will take a look!

danielinux, 2020-09-30:
@hckuo the example provided has now been merged to our wolfBoot-examples repository. Feel free to contact us again if you need more information on your use case.

Thanks!

-- 
D.

closed 2020-09-30 by danielinux