wolfSSL/wolfssl #3322: question /feature request : how to use cutsom rng or system rng only

cryptgithub, 2020-09-22:
i was looking through crypt libs and testing each one for performance and stability 
i saw most of them have abiilty to use system rng only (/dev/urandom on unix or cryptgenrandom on win32)
or even give interface your own rng into system (i understand it's more diffcult in c rather than OOL c++libs )

i didn't found any thing related to custom rng in documentation or in glace over random.c

question : how can we do any one them so wc functions accept our custom rng/system rng as parameter)

why : parameters  is giving me 5gb/s speed and wc default rng 150mb/s, also system rng may be more secure (even if not secure ,wc rng is seeded from it so makes wc rng weak too)

thanks for support

cryptgithub, 2020-09-22:
as i understand through reading source 
i just need to define CUSTOM_RAND_GENERATE_BLOCK marco and int CUSTOM_RAND_GENERATE_BLOCK(byte *output,int sz) function and fill output with what ever i want (output of os rng for example )
in no seed case (eg systemrng ) don't need even CUSTOM_RAND_GENERATE_SEED

any bugs,security/other problems with that ?
do i need to define anything else?

if that's the way , it would be better to add to documentation too

dgarske, 2020-09-22:
Hi @cryptgithub , 

Thanks for the questions regarding our RNG. We ensure our defaults lean towards more secure rather than tuned to random performance. 

To clarify your question is how to disable our PRNG (NIST Compliant DRBG) and use either a custom random source or the system random source directly?

Here are your random options:

1. Use a custom source with no PRNG:

```
#define WC_NO_HASHDRBG

/* Bypass P-RNG and use only HW RNG */
extern int my_rng_gen_block(unsigned char* output, unsigned int sz);
#undef  CUSTOM_RAND_GENERATE_BLOCK
#define CUSTOM_RAND_GENERATE_BLOCK  my_rng_gen_block
```

2. Use PRNG and define your own custom seed:

```
#define HAVE_HASHDRBG /* on by default */

extern int my_rand_gen_seed(byte* output, word32 sz);
#define CUSTOM_RAND_GENERATE_SEED  my_rand_gen_seed
```

3. Use PRNG with default seed sources ported to over 25 platforms. Including use of `/dev/random` and `/dev/urandom`

4. Use the Intel RDRAND or Intel RDSEED directly. Those are enabled with `HAVE_INTEL_RDRAND` or `HAVE_INTEL_RDSEED`.

5. For testing only you can use `WOLFSSL_GENSEED_FORTEST` to define a fixed seed (incrementing value).

Let me know if you have any issues or further questions.

Thanks,
David Garske, wolfSSL

cryptgithub, 2020-09-26:
thanks for kind replay 
please don't forget the bug i mentioned on windows builds it can simply be fixed by adding && in #ifdef USE_WINDOWS_API 
i close this issue

closed 2020-09-26 by cryptgithub