wolfSSL/wolfssl #3329: wolfSSL_X509_check_host() broken (or behavior change) in 4.5.0

vuvova, 2020-09-23:
See the commit d1a82589f9c8f9a8f431 

Before that commit if `match` in `CheckForAltNames` was 0, then `CheckForAltNames` would return 1 and `CheckHostName` **would not** return `DOMAIN_NAME_MISMATCH`

After that commit  if `match` in `CheckForAltNames` is 0, then `CheckForAltNames` would return 0 and `CheckHostName` **would** return `DOMAIN_NAME_MISMATCH`

This broke tests in MariaDB that test host name validation (via `wolfSSL_X509_check_host`)

vuvova, 2020-09-23:
Just to comment on "what behavior is correct", the test succeeds with OpenSSL and 4.4.0, but fails with 4.5.0

dgarske, 2020-09-24:
Hi @vuvova ,

I see what you mean about the `return (match != -1)` vs `return match`. There is a PR #3193 post v4.5.0 that I believe resolves this.

If that does not resolve it can you provide the certificate and domain match expected to pass/fail?

Thanks, 
David Garske, wolfSSL

vuvova, 2020-09-24:
Yes, I've tried the master branch — the test passed. Thanks!

closed 2020-09-24 by vuvova