wolfSSL/wolfssl #3334: Possible EVP_CIPHER_CTX_init memory bug

bgermann, 2020-09-24:
I have found a possible memory bug that can be triggered by running SWUpdate's unit tests (with WOLFSSL_APACHE_HTTPD undefined). The bug description is available at https://github.com/sbabic/swupdate/commit/ba8c4a8bff622. I guess that EVP_CIPHER_CTX_init is the offending function.

dgarske, 2020-09-24:
Hi @bgermann for the report. 

I believe this issue has been fixed in https://github.com/wolfSSL/wolfssl/pull/3317

Thanks,
David Garske, wolfSSL

bgermann, 2020-09-24:
I just tested the current master and the problem is still there.

dgarske, 2020-09-24:
Hi @bgermann ,

Would you mind seeing if this PR https://github.com/wolfSSL/wolfssl/pull/3321 resolves it?

If not can you provide instructions for reproducing?

Thanks,
David Garske, wolfSSL

bgermann, 2020-09-24:
The commit message has instructions for reproducing: make a swupdate config with image encryption enabled, change the code not to contain the WOLFSSL_APACHE_HTTPD definition, make, make test

bgermann, 2020-09-24:
And activate wolfSSL as SSL provider!

bgermann, 2020-09-24:
You will see the fail with:
test_crypt: malloc.c:2379: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

bgermann, 2020-09-24:
The PR does not fix it as well. You can use the following defonfig on SWUpdate's master branch:
```
# CONFIG_SCRIPTS is not set
# CONFIG_MTD is not set
# CONFIG_LUA is not set
CONFIG_EXTRA_CFLAGS="-g"
CONFIG_DEBUG=y
CONFIG_BOOTLOADER_NONE=y
CONFIG_SSL_IMPL_WOLFSSL=y
CONFIG_HASH_VERIFY=y
CONFIG_ENCRYPTED_IMAGES=y
# CONFIG_LIBCONFIG is not set
```

dgarske, 2020-09-24:
Hi @bgermann ,

I spent some time looking over the swupdate project. This project is new to me and I am happy to see you added wolfSSL support (including PKCS 11). I had some trouble building on the Mac, but got through most of it (I can send you patches for that if you are interested).

The issue appears to be because the wolfSSL library build options are not being included in your project. So the size of the `EVP_CIPHER_CTX` structure between build-time and run-time are different.

Fixes pushed to a branch here:
https://github.com/dgarske/swupdate/commit/36e247479097abbeda19afa723269b13accdc93d

For reference the issue you were seeing was because our default openssl compatibility layer version number was using different API's based on the versions. Either it calls `EVP_CIPHER_CTX_init` or (`EVP_CIPHER_CTX_new` and `EVP_CIPHER_CTX_reset`).

Thanks,
David Garske, wolfSSL

bgermann, 2020-09-24:
I see. Thanks for the explanation and taking the time for it!

closed 2020-09-24 by bgermann