wolfSSL/wolfssl #3347: [C++] missing min function

pustekuchen91, 2020-09-29:
compiling with xc32\v2.41\bin\xc32-g++.exe leads to the following error:

```
build/*/production/_ext/1664057780/md5.o: In function `wc_Md5Update':
c:/git/pic32/*/firmware/src/third_party/wolfssl/wolfssl/wolfcrypt/src/md5.c:369: undefined reference to `min'
build/*/production/_ext/1664057780/aes.o: In function `wc_AesCfbDecrypt':
c:/git/pic32/*/firmware/src/third_party/wolfssl/wolfssl/wolfcrypt/src/aes.c:7168: undefined reference to `min'
build/*/production/_ext/1664057780/sha.o: In function `wc_ShaUpdate':
c:/git/pic32/*/firmware/src/third_party/wolfssl/wolfssl/wolfcrypt/src/sha.c:555: undefined reference to `min'
```
workaround: 
Define macros
```
#define	max(a,b)	(((a) > (b)) ? (a) : (b))
#define	min(a,b)	(((a) < (b)) ? (a) : (b))
```
should this handled by wolfssl library?

dgarske, 2020-09-29:
Hi @pustekuchen91 ,

We provide `min` / `max` implementations in our `wolfcrypt/src/misc.c` file here:
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/misc.c#L249

Make sure you do not have `WOLFSSL_HAVE_MIN` defined. Also make sure you are building `wolfcrypt/src/misc.c`. Note that file is for inlined functions, so the use of `NO_INLINE` will determine if its included or if it needs to be built on its own.

Thanks,
David Garske, wolfSSL

pustekuchen91, 2020-09-30:
thank you very much for the fast response. With your hint i was able to identify the problem.

I'm using the harmony framework, which defined `WOLFSSL_HAVE_MIN` and `WOLFSSL_HAVE_MAX` in configuration.h.

dgarske, 2020-09-30:
Hi Marc (@pustekuchen91),

Thanks for the followup and am glad you were able to fixe the issue.

That section in config.h should be wrapped in `#ifndef __cplusplus`. I'll make note of this and let Microchip know.

Thanks,
David Garske, wolfSSL

closed 2020-09-30 by dgarske

pustekuchen91, 2020-10-02:
Hi David,

thank you. Just a small addition which you could please forward to microchip to this topic. 
the file `misc.c` is added in the wrong logical folder of the project file `Project.X/nbproject/configurations.xml`
`misc.c is` added as externalFiles, which wont be compiled and still lead to the error.
```
    <logicalFolder name="ExternalFiles"
                   displayName="Important Files"
                   projectFiles="false">
      <logicalFolder name="f1" displayName="wolfcrypt" projectFiles="true">
        <itemPath>../src/third_party/wolfssl/wolfssl/wolfcrypt/src/misc.c</itemPath>
      </logicalFolder>
    </logicalFolder>
```

right would be to add `misc.c` to the logical folder `SourceFiles/woflcrypt`

PIC32MX795F512L is used here.

with kind regards,
Marc