wolfSSL/wolfssl #3370: Current master fails to compile

mouse07410, 2020-10-09:
MacOS Catalina 10.15.7, Xcode-12.0.1.
```sh
$ ./configure --prefix=/opt/local --enable-aesccm --enable-aesctr --enable-aesofb --enable-aescfb --enable-aesni --enable-camellia --enable-md2 --enable-blake2 --enable-blake2s --enable-sessioncerts --enable-keygen --enable-certgen --enable-hkdf --enable-ecccustcurves --enable-compkey --enable-curve25519 --enable-ed25519 --enable-curve448 --enable-ed448 --enable-fpecc --enable-eccencrypt --enable-psk --enable-psk-one-id --enable-stacksize --enable-anon --enable-idea --enable-cmac --enable-xts --enable-webserver --enable-webclient --enable-hc128 --enable-rabbit --enable-ocsp --enable-ocspstapling2 --enable-pkcs7 --enable-jni --enable-pwdbased --enable-enckeys --enable-scrypt --with-libz --enable-dtls --enable-eccshamir --enable-x963kdf --enable-rsapss --enable-pkcs11 --enable-aeskeywrap --with-ntru=/opt/local --enable-qsh
.  .  .  .  .
$ make
.  .  .  .  .
wolfcrypt/test/test.c:27544:5: error: use of undeclared identifier 'enveloped'
    enveloped = (byte *)XMALLOC(PKCS7_BUF_SIZE, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
    ^
wolfcrypt/test/test.c:27546:12: error: use of undeclared identifier 'enveloped'
    if ((! enveloped) || (! decoded)) {
           ^
wolfcrypt/test/test.c:27610:9: error: use of undeclared identifier 'enveloped'
    if (enveloped)
        ^
wolfcrypt/test/test.c:27611:15: error: use of undeclared identifier 'enveloped'
        XFREE(enveloped, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
              ^
4 errors generated.
make[2]: *** [Makefile:5419: wolfcrypt/test/testsuite_testsuite_test-test.o] Error 1
make[2]: *** Waiting for unfinished jobs....
wolfcrypt/test/test.c:27544:5: error: use of undeclared identifier 'enveloped'
    enveloped = (byte *)XMALLOC(PKCS7_BUF_SIZE, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
    ^
wolfcrypt/test/test.c:27546:12: error: use of undeclared identifier 'enveloped'
    if ((! enveloped) || (! decoded)) {
           ^
wolfcrypt/test/test.c:27610:9: error: use of undeclared identifier 'enveloped'
    if (enveloped)
        ^
wolfcrypt/test/test.c:27611:15: error: use of undeclared identifier 'enveloped'
        XFREE(enveloped, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
              ^
4 errors generated.
```

Complete configure output: [config-out.txt](https://github.com/wolfSSL/wolfssl/files/5351904/config-out.txt)

Complete make output: [make-out.txt](https://github.com/wolfSSL/wolfssl/files/5351906/make-out.txt)

mouse07410, 2020-10-09:
Fix: 
```diff
diff --git a/wolfcrypt/test/test.c b/wolfcrypt/test/test.c
index 5751d900f..62844d587 100644
--- a/wolfcrypt/test/test.c
+++ b/wolfcrypt/test/test.c
@@ -25645,12 +25645,12 @@ static int pkcs7enveloped_run_vectors(byte* rsaCert, word32 rsaCertSz,
                                       byte* eccCert, word32 eccCertSz,
                                       byte* eccPrivKey,  word32 eccPrivKeySz)
 {
-    int ret = 0, testSz, i;
-    int envelopedSz, decodedSz;
+    int ret = 0, testSz = 0, i = 0;
+    int envelopedSz = 0, decodedSz = 0;
 
-    byte   *enveloped;
-    byte   *decoded;
-    PKCS7* pkcs7;
+    byte   *enveloped = NULL;
+    byte   *decoded = NULL;
+    PKCS7* pkcs7 = NULL;
     WC_RNG rng;
 #ifdef PKCS7_OUTPUT_TEST_BUNDLES
     XFILE  pkcs7File;
@@ -27519,11 +27519,12 @@ typedef struct {
 static int pkcs7compressed_test(void)
 {
     int ret = 0;
-    int i, testSz;
-    int compressedSz, decodedSz;
-    PKCS7* pkcs7;
-    byte  *compressed;
-    byte  *decoded;
+    int i = 0, testSz = 0;
+    int compressedSz = 0, decodedSz = 0;
+    PKCS7* pkcs7 = NULL;
+    byte  *compressed = NULL;
+    byte  *decoded = NULL;
+    byte  *enveloped = NULL;
 #ifdef PKCS7_OUTPUT_TEST_BUNDLES
     XFILE pkcs7File;
 #endif
```

P.S. Leaving pointers uninitialized is not good. Same with integers.

kaleb-himes, 2020-10-09:
@ejohnstown please assign as a maintenance item.

ejohnstown, 2020-10-09:
This is related to #3371, it is the same issue. And someone is looking at that now.

closed 2020-10-09 by ejohnstown