wolfSSL/wolfssl #3371: Tests fail on MacOS

mouse07410, 2020-10-09:
MacOS Catalina 10.15.7, Xcode-12.0.1.

Configuration:
```
./configure --prefix=/opt/local --enable-aesccm --enable-aesctr --enable-aesofb --enable-aescfb --enable-aesni --enable-camellia --enable-md2 --enable-blake2 --enable-blake2s --enable-sessioncerts --enable-keygen --enable-certgen --enable-hkdf --enable-ecccustcurves --enable-compkey --enable-curve25519 --enable-ed25519 --enable-curve448 --enable-ed448 --enable-fpecc --enable-eccencrypt --enable-psk --enable-psk-one-id --enable-stacksize --enable-anon --enable-idea --enable-cmac --enable-xts --enable-webserver --enable-webclient --enable-hc128 --enable-rabbit --enable-ocsp --enable-ocspstapling2 --enable-pkcs7 --enable-jni --enable-pwdbased --enable-enckeys --enable-scrypt --with-libz --enable-dtls --enable-eccshamir --enable-x963kdf --enable-rsapss --enable-pkcs11 --enable-aeskeywrap --with-ntru=/opt/local --enable-qsh
```

Test failure:
```
PASS: scripts/openssl.test
PASS: scripts/external.test
PASS: scripts/resume.test
PASS: scripts/pkcallbacks.test
PASS: scripts/crl-revoked.test
PASS: scripts/psk.test
PASS: scripts/tls13.test
FAIL: testsuite/testsuite.test
./build-aux/test-driver: line 107:  4113 Abort trap: 6           "$@" > $log_file 2>&1
FAIL: scripts/unit.test
PASS: scripts/google.test
FAIL: scripts/ocsp.test
PASS: scripts/ocsp-stapling2.test
============================================================================
Testsuite summary for wolfssl 4.5.0
============================================================================
# TOTAL: 12
# PASS:  9
# SKIP:  0
# XFAIL: 0
# FAIL:  3
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
```

More specific:
```
.  .  .  .  .
COMPRESS test passed!
PKCS7encrypted  test passed!
PKCS7compressed test failed!
 error = -12101
stack used = 106000
FAIL testsuite/testsuite.test (exit status: 255)
.  .  .  .  .
```

Complete `test-suite.log`: [test-suite.log.txt](https://github.com/wolfSSL/wolfssl/files/5352042/test-suite.log.txt)

kaleb-himes, 2020-10-09:
@ejohnstown, please assign as a maintenance item.

ejohnstown, 2020-10-09:
@mouse07410, which commit of wolfSSL are you using?

ejohnstown, 2020-10-09:
Recreate on commit 4d11e3c with:

    $ ./configure --enable-pkcs7 --with-libz&&make&&./wolfcrypt/test/testwolfcrypt
    PKCS7compressed test failed!
     error = -12101

mouse07410, 2020-10-09:
> @mouse07410, which commit of wolfSSL are you using?

The `master` that was current at the issue opening date and time. And the 4.5.0 release, if memory serves.

ejohnstown, 2020-10-09:
Also, not a Mac thing. I recreated the issue under Ubuntu.

douzzer, 2020-10-16:
These issues should be cleared up on all platforms by #3377 and #3395, both merged.

closed 2020-10-16 by douzzer