wolfSSL/wolfssl #3376: Tests fail...

mouse07410, 2020-10-09:
```
PASS: scripts/openssl.test
PASS: scripts/external.test
PASS: scripts/sniffer-testsuite.test
PASS: scripts/psk.test
PASS: scripts/resume.test
PASS: scripts/crl-revoked.test
PASS: scripts/pkcallbacks.test
PASS: scripts/tls13.test
PASS: scripts/google.test
FAIL: scripts/ocsp.test
FAIL: scripts/ocsp-stapling.test
PASS: scripts/ocsp-stapling-with-ca-as-responder.test
PASS: scripts/ocsp-stapling2.test
FAIL: testsuite/testsuite.test
./build-aux/test-driver: line 107:  6121 Abort trap: 6           "$@" > $log_file 2>&1
FAIL: scripts/unit.test
============================================================================
Testsuite summary for wolfssl 4.5.0
============================================================================
# TOTAL: 15
# PASS:  11
# SKIP:  0
# XFAIL: 0
# FAIL:  4
# XPASS: 0
# ERROR: 0
============================================================================
See ./test-suite.log
Please report to https://github.com/wolfssl/wolfssl/issues
============================================================================
```

Something looks strange. From running `scripts/unit.test` manually:
```
.  .  .  .  .
   wolfSSL_X509_PUBKEY_get(): passed
   wolfSSL_sk_CIPHER_description:
ERROR - tests/api.c line 32168 failed with:
    expected: test_str != badStr
    result:   "unknown" == "unknown"

Abort trap: 6
```

Complete log: [test-suite.log.txt](https://github.com/wolfSSL/wolfssl/files/5356601/test-suite.log.txt)

Config:
```
Configuration summary for wolfssl version 4.5.0

   * Installation prefix:        /opt/local
   * System type:                apple-darwin19.6.0
   * Host CPU:                   x86_64
   * C Compiler:                 clang
   * C Flags:                    -O3 -std=gnu18 -march=native -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk   -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wshorten-64-to-32 -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:               clang++
   * C++ Flags:                  -std=gnu++2a -O3 -march=native -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk
   * CPP Flags:                   -DHAVE_NTRU -DHAVE_TLS_EXTENSIONS -I/opt/local/include
   * CCAS Flags:                 -O3 -std=gnu18 -march=native -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk  
   * LIB Flags:                   -pie -Werror 
   * Debug enabled:              no
   * Coverage enabled:           
   * Warnings as failure:        yes
   * make -j:                    21
   * VCS checkout:               yes

   Features 
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          yes
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * fastmath:                   yes
   * Assembly Allowed:           yes
   * sniffer:                    yes
   * snifftest:                  yes
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     yes
   * AES-CBC:                    yes
   * AES-GCM:                    yes
   * AES-CCM:                    yes
   * AES-CTR:                    yes
   * DES3:                       no
   * IDEA:                       no
   * Camellia:                   yes
   * NULL Cipher:                no
   * MD5:                        yes
   * RIPEMD:                     no
   * SHA:                        yes
   * SHA-224:                    yes
   * SHA-384:                    yes
   * SHA-512:                    yes
   * SHA3:                       yes
   * SHAKE256:                   yes
   * BLAKE2:                     yes
   * CMAC:                       yes
   * keygen:                     yes
   * certgen:                    yes
   * certreq:                    yes
   * certext:                    yes
   * certgencache:               yes
   * HC-128:                     yes
   * RABBIT:                     yes
   * CHACHA:                     yes
   * Hash DRBG:                  yes
   * PWDBASED:                   yes
   * scrypt:                     yes
   * wolfCrypt Only:             no
   * HKDF:                       yes
   * X9.63 KDF:                  yes
   * MD4:                        yes
   * PSK:                        yes
   * Poly1305:                   yes
   * LEANPSK:                    no
   * LEANTLS:                    no
   * RSA:                        yes
   * RSA-PSS:                    yes
   * DSA:                        no
   * DH:                         yes
   * DH Default Parameters:      yes
   * ECC:                        yes
   * ECC Custom Curves           yes
   * CURVE25519:                 yes
   * ED25519:                    yes
   * CURVE448:                   yes
   * ED448:                      yes
   * FPECC:                      yes
   * ECC_ENCRYPT:                yes
   * ASN:                        yes
   * Anonymous cipher:           yes
   * CODING:                     yes
   * MEMORY:                     yes
   * I/O POOL:                   no
   * LIGHTY:                     no
   * HAPROXY:                    no
   * STUNNEL:                    no
   * Apache httpd:               no
   * NGINX:                      no
   * ASIO:                       yes
   * LIBWEBSOCKETS:              no
   * Qt                          no
   * Qt Unit Testing             no
   * SIGNAL:                     no
   * ERROR_STRINGS:              yes
   * DTLS:                       yes
   * SCTP:                       no
   * Indefinite Length:          no
   * Multicast:                  no
   * SSL v3.0 (Old):             no
   * TLS v1.0 (Old):             no
   * TLS v1.1 (Old):             yes
   * TLS v1.2:                   yes
   * TLS v1.3:                   yes
   * Post-handshake Auth:        no
   * Early Data:                 no
   * Send State in HRR Cookie:   no
   * OCSP:                       yes
   * OCSP Stapling:              yes
   * OCSP Stapling v2:           yes
   * CRL:                        yes
   * CRL-MONITOR:                yes
   * Persistent session cache:   yes
   * Persistent cert    cache:   yes
   * Atomic User Record Layer:   yes
   * Public Key Callbacks:       yes
   * NTRU:                       yes
   * QSH:                        yes
   * Whitewood netRandom:        no
   * Server Name Indication:     yes
   * ALPN:                       yes
   * Maximum Fragment Length:    no
   * Trusted CA Indication:      no
   * Truncated HMAC:             yes
   * Supported Elliptic Curves:  yes
   * FFDHE only in client:       no
   * Session Ticket:             no
   * Extended Master Secret:     yes
   * Renegotiation Indication:   no
   * Secure Renegotiation:       no
   * Fallback SCSV:              no
   * All TLS Extensions:         no
   * PKCS#7                      yes
   * wolfSSH                     no
   * wolfSCEP                    no
   * Secure Remote Password      yes
   * Small Stack:                no
   * Linux Kernel Module:        no
   * valgrind unit tests:        no
   * LIBZ:                       no
   * Examples:                   yes
   * Crypt tests:                yes
   * Stack sizes in tests:       yes
   * User Crypto:                no
   * Fast RSA:                   no
   * Single Precision:           no
   * Async Crypto:               no
   * PKCS#11:                    yes
   * PKCS#12:                    yes
   * Cavium Nitox:               no
   * Cavium Octeon (Sync):       no
   * Intel Quick Assist:         no
   * ARM ASM:                    no
   * AES Key Wrap:               yes
   * Write duplicate:            no
   * Xilinx Hardware Acc.:       no
   * Inline Code:                yes
   * Linux AF_ALG:               no
   * Linux devcrypto:            no
   * Crypto callbacks:           yes

---
```

ejohnstown, 2020-10-09:
@mouse07410 , can you please post your configure command line? Thanks!

ejohnstown, 2020-10-09:
I'm going to assume the other configure command options. And I think I know which combination is failing. I should have a fix soon.

ejohnstown, 2020-10-09:
And that combination didn't fail.

If you can post the output of `./config.status --config` that would be helpful. Thank you.

mouse07410, 2020-10-09:
```
$ ./config.status --config
'--prefix=/opt/local' '--enable-aesccm' '--enable-aesctr' '--enable-aesofb' '--enable-aescfb' '--enable-aesni' '--enable-aeskeywrap' '--enable-asio' '--enable-sniffer' '--enable-srp' '--enable-truncatedhmac' '--enable-camellia' '--enable-blake2' '--enable-blake2s' '--enable-sessioncerts' '--enable-keygen' '--enable-certgen' '--enable-certreq' '--enable-certgencache' '--enable-testcert' '--enable-md4' '--enable-md5' '--enable-examples' '--enable-crypttests' '--enable-hkdf' '--enable-ecccustcurves' '--enable-compkey' '--enable-curve25519' '--enable-ed25519' '--enable-curve448' '--enable-ed448' '--enable-fpecc' '--enable-eccencrypt' '--enable-psk' '--enable-psk-one-id' '--enable-stacksize' '--enable-anon' '--enable-cmac' '--enable-xts' '--enable-webserver' '--enable-webclient' '--enable-hc128' '--enable-rabbit' '--enable-ocsp' '--enable-ocspstapling' '--enable-ocspstapling2' '--enable-jni' '--enable-pwdbased' '--enable-enckeys' '--enable-scrypt' '--enable-dtls' '--enable-eccshamir' '--enable-x963kdf' '--enable-rsapss' '--enable-pkcs7' '--enable-pkcs11' '--enable-pkcs12' '--with-ntru=/opt/local' '--enable-qsh' 'CC=clang' 'CFLAGS=-O3 -std=gnu18 -march=native -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk'
```

ejohnstown, 2020-10-09:
The smallest possible configuration to recreate:

    $ ./configure --enable-opensslall --with-ntru=./NTRU --enable-qsh

If you leave out the QSH option, it doesn't fail.

mouse07410, 2020-10-09:
Interesting. Though I must say that I cannot leave out the QSH. ;-)

ejohnstown, 2020-10-09:
I was just documenting the simplest configuration to recreate the issue. Also, you get the same error if you enable secure renegotiation indication. Same cause.

PR #3379 (2020-10-12): Maintenance Fixes
1. Add1. The test_wolfSSL_X509V3_EXT_print() test was using stderr for output, changed to stdout.
2. A call to XFREAD wasn't typecasting its output to the size of the variable getting the output in decodedCertCache_test(). (fixes #3375)
3. Improve the reporting of the NTRU based cipher suites with the function wolfSSL_sk_CIPHER_description().
4. When building the list of ciphers with wolfSSL_get_ciphers_compat(), skip the fake indicator ciphers like the renegotiation indication and the quantum-safe hybrid since they do not have encryption or mac algorithms associated to them. [... cut]
closed 2020-10-12 by dgarske