wolfSSL/wolfssl #3412: Compilation error in 4.5.0 code with -Wall and -O3

vppillai, 2020-10-22:
While compiling 4.5.0 code with XC32 2.50 PRO compiler with `-Wall` and `-O3` , the following error is thrown in code where TLS1.3 is disabled. 

```
../src/third_party/wolfssl/src/internal.c: In function 'ProcessPeerCerts':
../src/third_party/wolfssl/src/internal.c:10309:8: error: assuming signed overflow does not occur when assuming that (X - c) > X is always false [-Werror=strict-overflow]
     if (args->certIdx > args->count)
        ^
```

kaleb-himes, 2020-10-23:
@elms, Can you look into this maintenance issue when you have a moment? Thanks!

- KH

elms, 2020-10-27:
@vppillai I'm working to reproduce this. While I'm getting that setup a few things may help in the meantime.
 
1. Can you run `xc32-gcc -v` and share the output?
1. Please provide the full command line being used for compiling internal.c?
1. Can you check if this occurs with `-O2`?

vppillai, 2020-11-03:
1. `xc32-gcc -v`
```
Microchip Language Tool Shell Version  (Build date: Aug 30 2020).
Copyright (c) 2012-2017 Microchip Technology Inc. All rights reserved
Using built-in specs.
COLLECT_GCC=C:\Program Files\Microchip\xc32\v2.50\bin\bin/pic32m-gcc.exe
COLLECT_LTO_WRAPPER=c:/program files/microchip/xc32/v2.50/bin/bin/gcc/pic32mx/4.8.3/lto-wrapper.exe
Target: pic32mx
Configured with: /data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/src/pic32m-gcc/gcc/configure --target=pic32mx --prefix=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw --program-prefix=pic32m- --with-sysroot=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/pic32mx --with-bugurl=http://www.microchip.com/support --host=x86_64-w64-mingw32 --build=x86_64-linux-gnu --bindir=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/bin/bin --infodir=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/share/doc/xc32-pic32m-gcc/info --mandir=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/share/doc/xc32-pic32m-gcc/man --libdir=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/lib --libexecdir=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/bin/bin --with-build-sysroot=/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/install-mingw/pic32mx --enable-stage1-languages=c --enable-languages=c,c++ --disable-hosted-libstdcxx --disable-libstdcxx-pch --disable-libstdcxx-verbose --disable-libssp --disable-libmudflap --disable-libffi --disable-libfortran --disable-bootstrap --disable-shared --disable-__cxa_atexit --disable-nls --disable-libgomp --disable-threads --disable-sim --disable-decimal-float --disable-libquadmath --disable-shared --disable-checking --disable-maintainer-mode --enable-lto --enable-fixed-point --enable-gofast --enable-static --enable-sgxx-sde-multilibs --enable-sjlj-exceptions --enable-poison-system-directories --enable-obsolete --enable-cxx-flags=-mno-smart-io --without-isl --without-cloog --without-headers --with-newlib --with-dwarf2 --with-gnu-as --with-gnu-ld CC_FOR_TARGET=pic32m-gcc CXX_FOR_TARGET=pic32m-g++ AS_FOR_TARGET=pic32m-as AR_FOR_TARGET=pic32m-ar RANLIB_FOR_TARGET=pic32m-ranlib 'XGCC_FLAGS_FOR_TARGET=-frtti -fexceptions -fno-enforce-eh-specs -mno-smart-io' 'CFLAGS_FOR_TARGET=-O2 -mno-smart-io -fno-short-double -fomit-frame-pointer -fshort-wchar' 'CXXFLAGS_FOR_TARGET=-O2 -mno-smart-io -fno-short-double -fomit-frame-pointer -fshort-wchar -frtti -fexceptions' CXX=x86_64-w64-mingw32-g++ CC=x86_64-w64-mingw32-gcc CXXFLAGS='-g -O2 -fpermissive' CPPFLAGS='-D_BUILD_MCHP_ -DSKIP_LICENSE_MANAGER -I/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/build-mingw/opt/include -D_BUILD_XC32_' LDFLAGS='-static -L/data01/build/bamboo/xml-data/build-dir/M32DEV-PT37-BMT/build-mingw/opt/lib'
Thread model: single
gcc version 4.8.3 MPLAB XC32 Compiler v2.50 (Microchip Technology)
```

1.  full command line for `internal.c`
```
"C:\Program Files\Microchip\xc32\v2.50\bin\xc32-gcc.exe"    -g -x c -c -mprocessor=32MZ1025W104132  -ffunction-sections -fdata-sections -ftoplevel-reorder -O3 -fschedule-insns -fschedule-insns2 -DWOLFSSL_IGNORE_FILE_WARN -DHAVE_CONFIG_H -I"../src/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MZ" -I"../src/third_party/rtos/FreeRTOS/Source/include" -I"../src/config/pic32mz_w1_curiosity/driver/wifi/pic32mzw1/include/" -I"../src/third_party/wolfssl/wolfssl" -I"../src/third_party/wolfssl" -I"../src" -I"../src/config/pic32mz_w1_curiosity" -I"../src/packs/PIC32MZ1025W104132_DFP" -I"../src/config/pic32mz_w1_curiosity/library" -I"../src/config/pic32mz_w1_curiosity/library/tcpip/src" -I"../src/config/pic32mz_w1_curiosity/library/tcpip/src/common" -I"../src/config/pic32mz_w1_curiosity/library/cryptoauthlib" -I"../src/config/pic32mz_w1_curiosity/library/cryptoauthlib/crypto" -I"../src/third_party/paho.mqtt.embedded-c" -I"../src/config/pic32mz_w1_curiosity/system/fs/fat_fs/file_system" -I"../src/config/pic32mz_w1_curiosity/system/fs/fat_fs/hardware_access" -Werror -Wall -MMD -MF "build/pic32mz_w1_curiosity/production/_ext/2046716756/internal.o.d" -o build/pic32mz_w1_curiosity/production/_ext/2046716756/internal.o ../src/third_party/wolfssl/src/internal.c    -DXPRJ_pic32mz_w1_curiosity=pic32mz_w1_curiosity  -no-legacy-libc    -minterlink-compressed -mdfp="C:/Users/i16658/.mchp_packs/Microchip/PIC32MZ-W_DFP/1.3.177"  
```

1. It does not happen with `-O2`

1. In case you want the project where this is happening, it is at https://github.com/MicrochipTech/PIC32MZW1_Curiosity_OOB

elms, 2022-05-16:
@vppillai I was not able to reproduce this issue locally. I'm going to close this issue as it is on an old version and I couldn't reproduce. If you still have any issues with v5.3.0, please open a new bug report.

closed 2022-05-16 by elms