wolfSSL/wolfssl #3428: ./configure --enable-sni does not enable SNI callbacks

gstrauss, 2020-10-27:
`./configure --enable-sni` does not enable SNI callbacks

(also `--enable-opensslall` should probably also set `--enable-sni`)

SNI-related code should wrapped in the preprocessor directive `#ifdef HAVE_SNI` (and maybe `#ifndef NO_WOLFSSL_SERVER`) and not the current inconsistent mess of different sets of preprocessor directives.  (Aside: `HAVE_SNI` would be better named as `WOLFSSL_HAVE_SNI`, and that pattern would ideally be applied to all options in `wolfssl/options.h`)

Here's a smattering of different sets of preprocessor directives:

wolfssl/internal.h
```
#if defined(OPENSSL_ALL) || (defined(OPENSSL_EXTRA) && (defined(HAVE_STUNNEL) || \
                             defined(WOLFSSL_NGINX) || defined(HAVE_LIGHTY) || \
                             defined(WOLFSSL_HAPROXY) || defined(WOLFSSL_OPENSSH) ))
    CallbackSniRecv sniRecvCb;
    void*           sniRecvCbArg;
#endif
```

src/internal.c  (note that `SNI_Callback()` is silently not called for some preprocessor cases where the callback has been set)
```
#ifndef NO_WOLFSSL_SERVER

...

    #if defined(OPENSSL_ALL) || defined(HAVE_STUNNEL) || defined(WOLFSSL_NGINX) || defined(WOLFSSL_HAPROXY)
                if((ret=SNI_Callback(ssl)))
                    goto out;
                ssl->options.side = WOLFSSL_SERVER_END;
    #endif

...

#if defined(OPENSSL_ALL) || defined(HAVE_STUNNEL) || defined(WOLFSSL_NGINX) || \
    defined(WOLFSSL_HAPROXY)
    int SNI_Callback(WOLFSSL* ssl)
    {
        ...
    }
#endif /* OPENSSL_ALL || HAVE_STUNNEL || WOLFSSL_NGINX || WOLFSSL_HAPROXY */

#endif /* NO_WOLFSSL_SERVER */
```

wolfssl/ssl.h
```
#if defined(OPENSSL_ALL) || defined(HAVE_STUNNEL) || defined(WOLFSSL_NGINX) \
    || defined(WOLFSSL_HAPROXY) || defined(OPENSSL_EXTRA) || defined(HAVE_LIGHTY)

...

WOLFSSL_API void wolfSSL_CTX_set_servername_callback(WOLFSSL_CTX *,
        CallbackSniRecv);
WOLFSSL_API int wolfSSL_CTX_set_tlsext_servername_callback(WOLFSSL_CTX *,
        CallbackSniRecv);

WOLFSSL_API int  wolfSSL_CTX_set_servername_arg(WOLFSSL_CTX *, void*);

...

#endif /* OPENSSL_ALL || HAVE_STUNNEL || WOLFSSL_NGINX || WOLFSSL_HAPROXY || HAVE_LIGHTY */
```

src/ssl.c
```
#if defined(OPENSSL_ALL) || (defined(OPENSSL_EXTRA) && (defined(HAVE_STUNNEL) || \
                             defined(WOLFSSL_NGINX) || defined(HAVE_LIGHTY) || \
                             defined(WOLFSSL_HAPROXY) || defined(WOLFSSL_OPENSSH)))

...

#ifdef HAVE_SNI

...

#ifndef NO_WOLFSSL_SERVER
const char * wolfSSL_get_servername(WOLFSSL* ssl, byte type)
{
    ...
}
#endif /* NO_WOLFSSL_SERVER */
#endif /* HAVE_SNI */

...

void wolfSSL_CTX_set_servername_callback(WOLFSSL_CTX* ctx, CallbackSniRecv cb)
{
    ...
}

int wolfSSL_CTX_set_tlsext_servername_callback(WOLFSSL_CTX* ctx,
                                               CallbackSniRecv cb)
{
    ...
}

int wolfSSL_CTX_set_servername_arg(WOLFSSL_CTX* ctx, void* arg)
{
    ...
}

...

#endif /* OPENSSL_ALL || (OPENSSL_EXTRA && (HAVE_STUNNEL || WOLFSSL_NGINX || HAVE_LIGHTY)) */
```

kaleb-himes, 2020-10-30:
@gstrauss,

Thank you so much for the suggestions. We very much appreciate and agree with your feedback. I will add it to the internal feature request list as an enhancement to:

A) Make flags more consistent - HAVE_SNI to WOLFSSL_HAVE_SNI
B) tie all SNI features together under the single common flag WOLFSS_HAVE_SNI regardless if some are only available under openssl compatibility.

Warm Regards,

K

gstrauss, 2020-10-30:
FreeBSD will not ship lighttpd mod_wolfssl because of this mess of processor directives.  The wolfssl package on FreeBSD is not built with `--enable-opensslall`.  I might have provided a patch here, except the last patch submission I made is 3 months old, after having been reviewed and validated months ago.

closed 2021-01-14 by haydenroche5