wolfSSL/wolfssl #3430: [ESP8266] Core 0 panic on wolfcrypt random.c function

mpietron, 2020-10-27:
Hi,
I use wolfssl in popular esp8266 Homekit application [https://github.com/maximkulkin/esp-homekit](https://github.com/maximkulkin/esp-homekit). I use make to compile application without any error. But when I run my app I have error in random function. Any solution for this problem? Wat I do wrong?

I will be grateful for any help

```
--- idf_monitor on /dev/cu.usbserial-0001 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (8439) tcpip_adapter: sta ip: 10.0.1.72, mask: 255.255.255.0, gw: 10.0.1.1
I (8441) wifi station: got ip:10.0.1.72
I (8443) wifi station: connected to ap SSID:*********** password:*************
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x200000
>>> HomeKit: Generated new accessory ID: 81:BE:83:DF:7F:9A
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 0 register dump:
PC      : 0x40270530  PS      : 0x00000030  A0      : 0x40270530  A1      : 0x3fff78e0
0x40270530: hwrand_generate_block at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:943
 (inlined by) wc_RNG_GenerateBlock at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:883

A2      : 0xc7f54ac8  A3      : 0xaa5126a2  A4      : 0x0000001e  A5      : 0x40107a6a
A6      : 0x40107a5c  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x0000ffff
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x40107a4a  A13     : 0x00000000
A14     : 0x00000004  A15     : 0x00000020  SAR     : 0x00000013  EXCCAUSE: 0x00000009

Backtrace: 0x40270530:0x3fff78e0 0x402655d4:0x3fff7900 0x4023afe1:0x3fff7910 0x40239c90:0x3fff7930 0x40239d00:0x3fff7940
0x40270530: hwrand_generate_block at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:943
 (inlined by) wc_RNG_GenerateBlock at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/random.c:883

0x402655d4: wc_ed25519_make_key at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/wolfssl/wolfssl-4.1.0/wolfcrypt/src/ed25519.c:713

0x4023afe1: crypto_ed25519_generate at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/homekit/src/crypto.c:363

0x40239c90: homekit_accessory_key_generate at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/homekit/src/server.c:3393

0x40239d00: homekit_server_task at /Users/michal/Programming/Embedded/Workspace/Eclipse/Espressif/esp8266/RTOS/homekit/components/homekit/src/server.c:3427

��*�P*V��ET����T��K��u�U+UZ�Z�T��T
                                  .......
                                  .......
                                  .......
                                  .......                                 
T������]UZJPqQQ��U�
I (464) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (467) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (536) phy_init: phy ver: 1163_0
I (556) wifi station: wifi_init_sta finished.
I (678) wifi:state: 0 -> 2 (b0)
I (683) wifi:state: 2 -> 3 (0)
I (693) wifi:state: 3 -> 5 (10)
I (703) wifi:connected with *************, aid = 4, channel 11, HT20, bssid = *********
I (2459) tcpip_adapter: sta ip: 10.0.1.72, mask: 255.255.255.0, gw: 10.0.1.1
I (2461) wifi station: got ip:10.0.1.72
I (2468) wifi station: connected to ap SSID:********* password:************
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 81:BE:83:DF:7F:9A
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 55 from 10.0.1.112
>>> homekit_client_process: [Client 55] Got 139 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 47600
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 47516
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 55] Initializing crypto
[... 44 lines cut ...]
>>> homekit_server_on_pair_setup: [Client 55] Calculating shared secret
>>> homekit_server_on_pair_setup: [Client 55] Decrypting payload
>>> homekit_server_on_pair_setup: [Client 55] Importing device public key
>>> homekit_server_on_pair_setup: [Client 55] Calculating DeviceX
>>> homekit_server_on_pair_setup: [Client 55] Verifying device signature
>>> HomeKit: Added pairing with 0A4ABC84-BBFF-4D88-87FB-AF95180AF556
>>> homekit_server_on_pair_setup: [Client 55] Exporting accessory public key
>>> homekit_server_on_pair_setup: [Client 55] Calculating AccessoryX
>>> homekit_server_on_pair_setup: [Client 55] Generating accessory signature
>>> homekit_server_on_pair_setup: Free heap: 44908
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 1 value (17 bytes): 81:BE:83:DF:7F:9A
>>> tlv_debug: Type 3 value (32 bytes): \xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF
>>> tlv_debug: Type 10 value (64 bytes): T\xD2\xF6\x8F\xAE\x18\x1E\x10\xB7;>i\x8A\x06\xE7W\xE9\xB0\xC2\x0E[S\x9B<D\xCBDG\xFCY.\xFDY)\xB7\xB0\xC9\xE8\xA4!\xBC\x00bt\xA6Z\xF1\xFB|\xD2\xBD\xAE\x94\xA2\xAB\x88\xFB\xD5\xF2\xD2+ly\x0F
>>> homekit_server_on_pair_setup: [Client 55] Encrypting response
>>> send_tlv_response: [Client 55] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x06
>>> tlv_debug: Type 5 value (135 bytes): .\xFE\xE2NkB\xB7\xF9\x12\x00F\x05\xC6u\xB6\xF5s\xF3"\xB9OW\xF9\x0E\xD6\xB8_J\xE5\x8E\xF2|\x928\x05B\xF5\x14O\xBA=\xEDk\xCB\x97\x84\xD3\xC6`y�r=\xB5)\x06d\xFB\\\x9A\xEDY\xE3\xE2\xF7\x0D\xC4\x02�\x95\x8A\xE7\x9E\xD4\xCE8\xB8Q\xF4\xF2\xCC\xBE\xEB\xF1\xC7\x91\xB2{\x02\x15�\xBD|I\xEF\x8F\xBD\x19\x1A\x9B~\xE3\x17\xCD\x0Cdr\xB6=\xDAV6L�\xB5\xD9\xC6\x1B\xC9\xBF!\x9A\x8B\xB9Y\xF5\xC6j\xBD(\x9BJ@\x1E$
>>> client_send: [Client 55] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 140\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x06\x05\x87.\xFE\xE2NkB\xB7\xF9\x12\x00F\x05\xC6u\xB6\xF5s\xF3"\xB9OW\xF9\x0E\xD6\xB8_J\xE5\x8E\xF2|\x928\x05B\xF5\x14O\xBA=\xEDk\xCB\x97\x84\xD3\xC6`y�r=\xB5)\x06d\xFB\\\x9A\xEDY\xE3\xE2\xF7\x0D\xC4\x02�\x95\x8A\xE7\x9E\xD4\xCE8\xB8Q\xF4\xF2\xCC\xBE\xEB\xF1\xC7\x91\xB2{\x02\x15�\xBD|I\xEF\x8F\xBD\x19\x1A\x9B~\xE3\x17\xCD\x0Cdr\xB6=\xDAV6L�\xB5\xD9\xC6\x1B\xC9\xBF!\x9A\x8B\xB9Y\xF5\xC6j\xBD(\x9BJ@\x1E$
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 55] Successfully paired
>>> homekit_client_process: [Client 55] Finished processing
>>> HomeKit: [Client 55] Closing client connection
```

dgarske, 2020-10-27:
Hi @mpietron ,

This looks like a stack issue. The default implementation of `ed25519` is a large code base. Have you defined `CURVED25519_SMALL` to use the small version?

You might also consider the build options. `WOLFSSL_GENERAL_ALIGNMENT 4` and `WOLFSSL_SMALL_STACK_CACHE`.

Also the v4.1.0 release is quite old. Is there a reason you haven't updated to the latest wolfSSL v4.5.0 release? 

Thanks,
David Garske, wolfSSL

dgarske, 2021-01-13:
Hi @mpietron ,

I wanted to check in and make sure you resolved this. Did the use of the small Ed/Curve25519 implementation help?

Thanks,
David Garske, wolfSSL

closed 2021-06-14 by dgarske