wolfSSL/wolfssl #3437: ssl->arrays->sessionIDSz is always 0, session can not resume correctly

neufeng, 2020-10-28:
When enable use session ticket with TLS1.2, in SetTicket just copy sessionID, but not fix sessionIDSz, this leads a bug when users want to reuse session.
Maybe below is a fix, I am not sure!

```
int SetTicket(WOLFSSL* ssl, const byte* ticket, word32 length)
{
    /* Free old dynamic ticket if we already had one */
    if (ssl->session.isDynamic) {
        XFREE(ssl->session.ticket, ssl->heap, DYNAMIC_TYPE_SESSION_TICK);
        ssl->session.ticket = ssl->session.staticTicket;
        ssl->session.isDynamic = 0;
    }

    if (length > sizeof(ssl->session.staticTicket)) {
        byte* sessionTicket =
                   (byte*)XMALLOC(length, ssl->heap, DYNAMIC_TYPE_SESSION_TICK);
        if (sessionTicket == NULL)
            return MEMORY_E;
        ssl->session.ticket = sessionTicket;
        ssl->session.isDynamic = 1;
    }
    ssl->session.ticketLen = (word16)length;

    if (length > 0) {
        XMEMCPY(ssl->session.ticket, ticket, length);
        if (ssl->session_ticket_cb != NULL) {
            ssl->session_ticket_cb(ssl,
                                   ssl->session.ticket, ssl->session.ticketLen,
                                   ssl->session_ticket_ctx);
        }
        /* Create a fake sessionID based on the ticket, this will
         * supersede the existing session cache info. */
        ssl->options.haveSessionId = 1;
#ifdef WOLFSSL_TLS13
        if (ssl->options.tls1_3) {
            XMEMCPY(ssl->session.sessionID,
                                 ssl->session.ticket + length - ID_LEN, ID_LEN);
        }
        else
#endif
        {
            XMEMCPY(ssl->arrays->sessionID,
                                 ssl->session.ticket + length - ID_LEN, ID_LEN);
            // fix session id size
            ssl->arrays->sessionIDSz = ID_LEN;
        }
    }

    return 0;
}
```

neufeng, 2020-10-30:
I find that 'GetDeepCopySession' is slow, wast about 100ms...... @kaleb-himes

kaleb-himes, 2020-11-04:
@ejohnstown please evaluate and assign as necessary.

SparkiDev, 2020-11-06:
Hi @neufeng,

I'm trying to understand what the bug is that this is fixing and I can't see it.
The client invalidates any session ID that was sent by the server.
My understanding is that it does not create a session ID when one wasn't sent.

Could you please expand on the bug you are seeing?

I'm not sure why you are seeing 100ms copy times but it may have something to with the locking.
Are there multiple threads?

neufeng, 2020-11-12:
I use wolf with curl, when enable session ticket, can not resume session.
In below code, you want to keep session id, but when session ticket is enable, ssl->arrays->sessionID is empty
```
/* free use of temporary arrays */
void FreeArrays(WOLFSSL* ssl, int keep)
{
    if (ssl->arrays) {
        if (keep) {
            /* keeps session id for user retrieval */
            XMEMCPY(ssl->session.sessionID, ssl->arrays->sessionID, ID_LEN);
            ssl->session.sessionIDSz = ssl->arrays->sessionIDSz;
        }
        if (ssl->arrays->preMasterSecret) {
            XFREE(ssl->arrays->preMasterSecret, ssl->heap, DYNAMIC_TYPE_SECRET);
            ssl->arrays->preMasterSecret = NULL;
        }
        XFREE(ssl->arrays->pendingMsg, ssl->heap, DYNAMIC_TYPE_ARRAYS);
        ssl->arrays->pendingMsg = NULL;
        ForceZero(ssl->arrays, sizeof(Arrays)); /* clear arrays struct */
    }
    XFREE(ssl->arrays, ssl->heap, DYNAMIC_TYPE_ARRAYS);
    ssl->arrays = NULL;
}
```

neufeng, 2020-11-12:
The copy time is my mistake! Why request time longer for the second time when request the same url?

SparkiDev, 2020-11-19:
Hi @neufeng,

I've put up a pull request with a proposed fix: https://github.com/wolfSSL/wolfssl/pull/3515
Instead of setting the a full length session ID, I've set a length one less that indicates that is fake/bogus.
When using session tickets, you should not be using the session ID to resume the session but the data in the session ticket.

Let me know if this works as you expect.

Requesting from the same URL should not take any longer.
If the first handshake is a full handshake then the second will be resumed if the session information or ticket is kept and be faster.

Can you tell me whether a full handshake is being performed in both cases?
If so, is there anything different about the messages being sent.
That is, are the same certificate being used, the same cipher suite used, the same key exchange and so on.

SparkiDev, 2021-11-15:
Hi @neufeng 

There has been changes to this code recently and I was wondering if this is still an issue for you?

Thanks

SparkiDev, 2022-01-11:
Hi @neufeng

Due to the long period of inactivity, I am closing this issue.

Thanks.

closed 2022-01-11 by SparkiDev