wolfSSL/wolfssl #3451: lots of syntax errors msvsc after upgrading to last clone of respository

cryptgithub, 2020-11-02:
i had older version working well 
cloned last version compiles well on default config (except one syntax error on chacha functions) but when defining USE_INTEL_SPEEDUP i get lots of syntex errors mostly for using ({ in Macros 

also without USE_INTEL_SPEEDUP , if you just have HAVE_CHACHA msvsc throws syntex error on chacha.c  wc_Chacha_purge_current_block function at   byte scratch[CHACHA_CHUNK_BYTES] = {};
msvsc don't accept it , if you want random data   byte scratch[CHACHA_CHUNK_BYTES] ; is fine 
if you want zeros  then  byte scratch[CHACHA_CHUNK_BYTES] = {0}; 

user_settings.h attached
[user_settings.h.txt](https://github.com/wolfSSL/wolfssl/files/5475228/user_settings.h.txt)

cryptgithub, 2020-11-02:
also the benchmark solution for msvsc at \wolfcrypt\benchmark can't be compiled for x64 when upgraded to msvsc 2019 
after lots of playing and googling couldn't fix , i had to create new project and import source files ...

thanks for your library

kaleb-himes, 2020-11-04:
@cryptgithub,

Can you tell us which MSVS solution file you were building if using one of our provided solution files?

If not can you provide a reproducer for our engineers to evaluate with your user_settings.h?

Thanks!

- K

cryptgithub, 2020-11-05:
for syntex errors of main wolfssl: to reproduce 
just clone the repository , open wolfssl64.sln in main directory 
compiles fine , open user_setting.h , change #if 0 to #if 1 , in  /* Optional Performance Speedups */
compiles fine
[image]

add #define USE_INTEL_SPEEDUP 
now we have lots of syntax erros , mostly for the way you have defined multi line marcos  (attached a screenshot)

recloned and you have resolved the  the  scratch[CHACHA_CHUNK_BYTES]={} ; error in wc_Chacha_purge_current_block()  secussfully 

reproduce benchmark x64 compiling issue: open benchmark.sln  in \wolfcrypt\benchmark using msvsc 2019 (must be 2019 )
you will be requested to confirm converation of sln fiel to last version , confirm 
you will see a html file opended in your browser , one of the lines tell us : <<	benchmark.vcproj: Attribute 'Detect64BitPortabilityProblems' of 'VCCLCompilerTool' is not supported in this version and has been removed during conversion.>>
now set benchamrk project in side soultion as defualt and try to compile 
for x86 compiles well , to try to change to x64 , wonn\t compile , try going to configation manger and adding x64, it won't allow you, this is a problem with converting fro mvery olf sln files , not just for your project 
the only way is to either create new project or edit 	benchmark.vcproj manully

to resolve this you most projects inlcuce one sln file for vs2006 and one for vs2013, or make the user use cmake to get visual studio solution file 

regards

cryptgithub, 2020-11-11:
@kaleb-himes  could you reproduce the errors ?

cryptgithub, 2020-11-13:
@douzzer  this syntax errors are resovled secussfully 
now we have issue #3489 
so i close this

closed 2020-11-13 by cryptgithub