wolfSSL/wolfssl #3460: Encrypt Aes-Gcm in chunks

cryptgithub, 2020-11-03:
i played a few hours with wc_AesGcmEncrypt_ex and wc_AesGcmSetKey and read the source of the functions 
couldn't get the same cipher encrypted in one step , when encrypting in chunks 
googled in it seem it's not possible and already there are requests , and there no documentation for EX function

yes this is free library and you can add/not add whatever you want 
your lib looks great in comparsion page in wikipedia not just embedded also for modern systems (like servers) , but i'm sure not many people will use your lib if there's not such **basic functions** for most used algorithms (aes-gcm) 
especially in embedded environments (witch seems to be your main target )  where there is small memory and can't load large buffers at once

hope you take my complaint for good  
regards

kaleb-himes, 2020-11-04:
@cryptgithub,

Can you checkout the following (which do include GCM support):

wolfSSL_EVP_CipherInit()
wolfSSL_EVP_CipherUpdate()
wolfSSL_EVP_CipherFinal()

Is this what you are looking for?

There are some examples of using the OpenSSL compatibility versions of those APIs' in wolfssl-root/tests/api.c also look for these test cases:

```
test_wolfssl_EVP_aes_gcm_AAD_2_parts()
test_wolfssl_EVP_aes_gcm()
```

Warm Regards,

- K

cryptgithub, 2020-11-05:
@kaleb-himes 
yes i cheked evp.c 
in function wolfSSL_EVP_CipherUpdate_GCM()
it seems the data i pass in cached by rellocating more buffer 
that means if i want to encrypt a stream of 8gb, it will allocate 4GB of Ram , witch putpurpose of wolfSSL_EVP_CipherUpdate under question, allocation of 8gb impossible on x86 compiled binary or embeded devices (witch it your libs main target )

i saw request for this future from 2018 in your forum  ,i think i can find a portable lib for gcm, and use the rest of wolfssl , but it's nice to include such basic futures in your lib 

i'd also appreciate if there's a way to get correct ciphere in chunked buffer , without getting correct tag ( i can calcuate tag with another libs in chunk , but speed of your gcm encryption is not easy to achieve )

regards

ejohnstown, 2022-12-30:
This is resolved at this point. While the EVP layer used to store all the data so that wc_AesGcmEncrypt() could deal with the updates, we have improved wc_AesGcmEncrypt() with the configure option aesgcm-stream. You can stream 2GB, easily, in random sized chunks, and it doesn't malloc() any memory.

closed 2022-12-30 by ejohnstown