wolfSSL/wolfssl #3484: How to get finished message

WKJay, 2020-11-11:
Is there any api to get the finished message like "SSL_get_finished" in openssl?

tmael, 2020-11-11:
Hi @WKJay 

Thanks for filing this issue. 

Can you tell us more about your project? 

SSL_get_finished and SSL_get_peer_finished functions are unavailable at the moment. AFIK, I don't think wolfSSL has the capability to allow applications to obtain the latest Finished messages sent to or expected from a peer. 
We'll be more than happy to add this to our OpenSSL compatibility expansion backlog. When do you need it? 

Best Regards,
Tesfa

WKJay, 2020-11-12:
Hi @tmael

Thanks for your reply.

I am recently working on  EST protocal. In this protocol, when I generate certificate signing request, I need to embed tls-unique information in it, which refers to the finished message. 

That's why I need this feature. But I am not in a hurry because I have found a way to obtain the latest finished message by modifying source code and it seems to work fine.

If this feature could be added in the future release, it will be very nice :)

Regards,
Jay

tmael, 2020-11-12:
Hi Jay,

Thanks for your project info! 
I'm glad you managed to obtain the Finished msg.  Would you mind sharing how you accomplished that? 

Thanks,
Tesfa

WKJay, 2020-11-12:
Hi Tesfa,

Of course I can share.

I found that wolfssl sends finished message use `SendFinished` function, which defined in `internal.c`. And in this funcion, it invokes `BuildFinished`  to build finished message. So I simply copy the buffer contains finished message to my own buffer after `BuildFinished`.

```C

/* make finished hashes */
    hashes = (Hashes*)&input[headerSz];
    ret = BuildFinished(ssl, hashes,
                     ssl->options.side == WOLFSSL_CLIENT_END ? client : server);
    if (ret != 0) return ret;
   
    XMEMCPY(finished_msg,hashes,TLS_FINISHED_SZ);  //THIS IS MY CODE 

```

I don't know if this operation is correct but I can acturally get what I want in this way .

Best Regards,
Jay

tmael, 2020-11-13:
@WKJay. Thanks for sharing. Looks good to me. 
BTW, feel free to send any questions to support@wolfssl.com.

WKJay, 2020-11-13:
@tmael Thanks for the reply :)

tmael, 2020-11-13:
You are welcome @WKJay!  I will go ahead and close out this issue. If any other questions come up, please do not hesitate to open a new issue or email support@wolfssl.com

closed 2020-11-13 by tmael