wolfSSL/wolfssl #3489: MSVSC , some of asm file not getting built and linked

cryptgithub, 2020-11-12:
after merging pull #3486 
and having #define USE_INTEL_SPEEDUP
when using lib (eg in benchmark project or test project)
we get unresolved external symbol reloated to avx2/avx codes for sha and gcm and chacha 
as i find out the prbolem in that the *.S asmbely files in wolfssl-master\wolfcrypt\src are not getting complied and linked in project except aes_asm.asm file 
[image]

i created separate issue because event with sha INTEL_SPEED ups that were fixed in last issue we have this issue for chacha and gcm

tmael, 2020-11-12:
Thanks for reporting this issue @cryptgithub!

cryptgithub, 2020-11-12:
@tmael  thanks for your replay and bug flag
masm can't parse GAS stype asm but
we can assemble the .S files with yasm with "yasm.exe -r nasm -p gas -f win64/win32 ..." 
the problem in we need to run c preprocessor in .S before passing to yasm  (just like gcc does)
i didn't find such an option in "custom build tool" setting of msvsc
but no np , we can do that by calling cl.exe /P  to get preproceesed file ,
but i think we need to write a sperate program to to get output of cl and pass to yasm and give to msvsc use the output when linking
or maybe there's much simpler solution i'm couldn't think of

tmael, 2020-11-13:
Hi @cryptgithub,

Can you try using the following settings and let me know if it works? 

```
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#undef USE_INTEL_SPEEDUP
```

cryptgithub, 2020-11-13:
@tmael
hi
yes that worked , as said aes_asm.asm is the only asm getting built because it's in intel asm syntax unlike other witch ate in at&t syntax 
and it has not c preproccessors marcos , so masm compiles it
so what we have hardware aes on windows but not sha/chacha/poly/sp(single precision computing) 

i think we're missing a .asm (intel syntax )file for .S files along in the project  and comments on the .S file tells 
we can simplly avoid having a intel syntax copy of the same asm code by removing c preprocessors marcos from .S files and using yasm in windows (many projects do this )

thanks for your effort

haydenroche5, 2021-01-15:
We are tracking this cleanup in our internal list of feature requests/maintenance items. Closing.

closed 2021-01-15 by haydenroche5