wolfSSL/wolfssl #3491: Connect to AWS IoT Core failed, error code -188

Shaing, 2020-11-12:
Hi,

I has surveyed for two days, and I am stuck.
I has tried below conditions
1. connect wo SNI
[image]

2. W/SNI
[image]

both of the result was -188, ASN no signer error to confirm failure.

Did I lost something? or has there successful experience can share some tip for me? or others issue? 

Thx,
A.H.

tmael, 2020-11-12:
Hi @Shaing,

Thanks for reaching out! 
I suspect the likely cause is a missing signer CA. 
Can you double-check that ./certs/AmazonRootCA1.pem is the Certificate Authority?

Does the X509v3 cert extensions contain the following?
```
$ openssl x509 -in ./certs/AmazonRootCA1.pem -noout -text
...
            X509v3 Basic Constraints: 
                CA:TRUE

```

Thanks,
Tesfa

Shaing, 2020-11-12:
Hi @tmael 

Thanks for reply fast.

and YES, here is following
```
$ openssl x509 -in AmazonRootCA.pem -noout -text
X509v3 extensions:
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Key Usage: critical
                Digital Signature, Certificate Sign, CRL Sign
```

tmael, 2020-11-12:
Hi @Shaing,

Thanks for the info!  But you were using AmazonRootCA1.pem, not AmazonRootCA.pem when launching your client application.

Shaing, 2020-11-12:
Hi @tmael,

It's same file but different name that I has confirmed.
Sorry for this confused.

Shaing, 2020-11-12:
Hi @tmael ,
If OK, could you try it again from AWS IoT Core(free) and analyse the root cause? It's very helpful for me and Embedded development.

Thank you so much.
A.H.

tmael, 2020-11-13:
Hi @Shaing,

I'm unsure where you got your `AmazonRootCA.pem`. 

I was able to connect with the following:
$ ./examples/client/client -h a21fz4ez46exvr-ats.iot.us-west-2.amazonaws.com -p 8883 -A certs/sf-class2-root.crt

`sf-class2-root.crt` was obtained by clicking the `here` link on this page:
https://aws.amazon.com/blogs/security/how-to-prepare-for-aws-move-to-its-own-certificate-authority/
* Note: Amazon doesn’t own this root and doesn’t have a test URL for it. The certificate can be downloaded from here.

Could you let me know if that works for you? 

Thanks

Shaing, 2020-11-16:
Hi @tmael ,

Yeah !!!
It works!!! Thank you so much.

and reply where is AmazonRootCA.pem from https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html
```
RSA 2048 bit key: Amazon Root CA 1.
```

The cert file info. is from AWS demo scenario.

Thx,
A.H.

tmael, 2020-11-16:
@Shaing, 
Glad it worked for you! 
I will go ahead and close out this issue. If any other questions come up, please do not hesitate to open a new issue or email support@wolfssl.com

closed 2020-11-16 by tmael