wolfSSL/wolfssl #3502: question :how to check if cureve25519 received publickey has vaild point

cryptgithub, 2020-11-15:
to avoid invaild curve attack  witch will simply expose our private key to attacker ,we should check if the point exist on our curve 
i checked wc_curve25519_check_public() function source 
it seems it just check if byte are zero or one or pubkey byte array has invaild size , but no checking if the point are on curve
many libs provide such a functionally 
how can we achieve that with wolfssl ?
if this is not possible i'd like to know have wolssl considered defense against other attacks on ecc and other algorithms 

thanks for support

SparkiDev, 2020-11-15:
Please see this website: http://cr.yp.to/ecdh.html#validate
"How do I validate Curve25519 public keys?"

The short answer is that the public key, when one ordinate, does not need to be validated for ECDH.

cryptgithub, 2020-11-17:
@SparkiDev my mistake, i didn't know cureve25519 differ in this fild from other curves
sorry for taking your time

closed 2020-11-17 by cryptgithub