wolfSSL/wolfssl #3517: How to unpad PKCS7 encrypt data?

zwxf, 2020-11-20:
Any API for unpad PKCS7 data? I have found this:[wc_PKCS7_PadData], But no more information about how to unpad it? Am i lost some information about it?
Hope for reply!

tmael, 2020-11-20:
Hi @zwxf,

Have you looked at the [PKCS7 API doc](https://www.wolfssl.com/doxygen/pkcs7_8h.html )?

zwxf, 2020-11-21:
> 
> 
> Hi @zwxf,
> 
> Have you looked at the [PKCS7 API doc](https://www.wolfssl.com/doxygen/pkcs7_8h.html)?
> You can also look at the [PKCS7 examples](https://github.com/wolfSSL/wolfssl-examples/tree/master/pkcs7)

thanks to reply, I see the API doc and example before.

tmael, 2020-12-11:
Hi @zwxf,

Did you find what you were looking for?

zwxf, 2020-12-11:
Hi @tmael,
I add this API to solve my problem:
`
int wc_PKCS7_UnPadData(byte* in, word32 inSz)
{
    word32 pad_sz = in[inSz - 1];

    if (pad_sz > inSz) {
        return -1;
    }

    return inSz - pad_sz;
}
`
Thanks to reply!

closed 2020-12-11 by zwxf

mirbeymercan, 2025-04-02:
```
int unpad_pkcs7(uint8_t *data, size_t data_len) {
	if (data_len == 0) {
		return 0; 
	}
	uint8_t padding_value = data[data_len - 1];
	if (padding_value > 16) {
		return -1;
	}

	return data_len - padding_value;
}
```