wolfSSL/wolfssl #3530: Build fails with clang --disable-fastmath

guidovranken, 2020-12-02:
If the compile is clang, and the --disable-fastmath argument is passed to the configure script, the following errors are printed:

```
wolfcrypt/src/aes.c:1653:25: error: unused function 'PreFetchTe' [-Werror,-Wunused-function]
static WC_INLINE word32 PreFetchTe(void)
                        ^
wolfcrypt/src/aes.c:2010:25: error: unused function 'PreFetchTd' [-Werror,-Wunused-function]
static WC_INLINE word32 PreFetchTd(void)
                        ^
wolfcrypt/src/aes.c:2026:25: error: unused function 'PreFetchTd4' [-Werror,-Wunused-function]
static WC_INLINE word32 PreFetchTd4(void)
                        ^
3 errors generated.
```

Tested on the latest wolfssl repository checkout.

tmael, 2020-12-03:
Hi @guidovranken,

Which version of clang and wolfSSL are you using? 
Looks like the issues have been fixed on the wolfSSL mainline. I was able to build fine with.  

```
$ CC=clang
$ ./configure --disable-fastmath
$ make
$ clang --version
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

```
Best Regards,
Tesfa

guidovranken, 2020-12-06:
Fixed by https://github.com/wolfSSL/wolfssl/pull/3526

closed 2020-12-06 by guidovranken