wolfSSL/wolfssl #3560: PKCS7signed     test failed!

WKJay, 2020-12-13:
Hi,

I ran the wolfcrypt_test and it printed "PKCS7signed     test failed!".

I looked into the code and found the failure was caused by comparing **sid** and **expectedSid** in test.c.

```c
    ret = wc_PKCS7_GetSignerSID(pkcs7, sid, &sidSz);
    if (ret != 0) {
        wc_PKCS7_Free(pkcs7);
        XFREE(sid, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
        return ret;
    }
    ret = XMEMCMP(sid, expectedSid, sidSz); /*<-------------*/
    XFREE(sid, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
    if (ret != 0) {
        wc_PKCS7_Free(pkcs7);
        return ret;
    }
```

It seems like the expectedSid was the original data in **client_cert_der_2048** but the sid is calculated by wc_Sha256Hash function with the expectedSid data. So they would never be the same.

Is there anything wrong here or my understanding is incorrect ? Please give me a hint.

Thanks,
Jay

tmael, 2020-12-15:
Hi @WKJay,

Thanks for contacting wolfSSL! 
Can you share your configurations and the version of wolfSSL you are using?
I'm not seeing the failure on Ubuntu 20.04.1 LTS with wolfSSL mainline code. 

```
$ cd $HOME/wolfssl
$ ./configure --enable-pkcs7 && make
$ ./wolfcrypt/test/testwolfcrypt
...
PKCS7encrypted  test passed!
PKCS7signed     test passed!
PKCS7enveloped  test passed!
PKCS7authenveloped  test passed!
...
```

WKJay, 2020-12-15:
Hi @tmael,

I've found the problem.

Because I defined **NO_SHA**, so it did a calculation instead of a direct memory copy  since the length was different from KEYID_SIZE.
```c
if (length == KEYID_SIZE) {
        XMEMCPY(cert->extAuthKeyId, input + idx, length);
    }
    else
        ret = CalcHashId(input + idx, length, cert->extAuthKeyId);
```
And then it caused the comparision failure.
```c
ret = XMEMCMP(sid, expectedSid, sidSz); 
    XFREE(sid, HEAP_HINT, DYNAMIC_TYPE_TMP_BUFFER);
    if (ret != 0) {
        wc_PKCS7_Free(pkcs7);
        return ret;
    }
```

After I removed NO_SHA, it passed the test. So it is my problem.

Thanks for your reply.

Best Regards,
Jay

tmael, 2020-12-15:
Hi @WKJay,

I'm glad you found the problem.  I will go ahead and close out this issue. If any other questions come up please do not hesitate to open a new issue anytime here or by emailing support@wolfssl.com.

Thanks,
Tesfa

closed 2020-12-15 by tmael